blueman-project/blueman #2140: Change question "Shall bluetooth get enabled automatically?"

bowstones, 2023-09-11:
blueman: 2.3.5
BlueZ: 5.66
Distribution: Ubuntu 23.04
Desktop environment: MATE 1.26.0

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

On starting Blueman for the first time, I am presented with the question, on its own, in a dialog box:

> **Shall bluetooth get enabled automatically**?

This question seems poorly phrased for two reasons:

1. The way it reads feels clunky to me. "Enable Bluetooth automatically?" would seem more natural (but would still be problematic; see the next point)
2. It's not clear what it's actually asking. Should Bluetooth be enabled automatically _when_? When I start Blueman? On login? After a system reboot?

I think it would be a good idea to improve the phrasing of this question.

infirit, 2023-09-11:
It's blueman that enables bluetooth when it starts. But it does a bit more and phrasing it properly without confusing users is a bit difficult.

Bluetooth adapter(s) were unpowerd by default system-wide, a design choice of the BlueZ maintainers. So blueman can power the adapter for the users.

However, it may do more than just power an adapter, it may also unblock the adapter's killswitch.

If you have suggestion how to better phrase this do let us know :-).

bowstones, 2023-09-12:
By "killswitch" I assume you mean when the adapter has been blocked using rfkill? If so, I think that it would be a good idea to mention this behaviour, as a user may think that Blueman will not automatically enable an adapter if they've done this. So my first thought is that the question should be phrased as something like "_Would you like your Bluetooth adapter to be enabled automatically when Blueman starts up? This will unblock the adapter if it has previously been blocked with rfkill._"

The problem is that not everyone will now what rfkill is (I didn't until now). Combined with the fact that a lot of users also won't know what Blueman is (distros often provide it, and it usually just appears as "Bluetooth" or similar in menus), that question would not be very user-friendly!  So yes, I see the problem..

On the other hand, the behaviour of overriding rfkill is not mentioned in the current phrasing, so it wouldn't be making anything worse to leave it out.

Maybe we could just have _Enable Bluetooth when Blueman starts up?"_. That solves the problem of not knowing when it is referring to, though there is still the issue of it sometimes not being actually clear what Blueman is.

infirit, 2023-09-12:
Yup, rfkill is what I meant.

We could add a question mark icon in the dialog which has a tooltip that explains what exactly happens. Not sure how many people will actually read it but then it's there for whomever is curious.

cschramm, 2023-10-29:
The string is gone since #1843

closed 2023-10-29 by cschramm