blueman-project/blueman #2141: src/device.c:set_wake_allowed_complete() Set device flags return status: Invalid Parameters

killertofus, 2023-09-16:
blueman: 2.4-alpha git version
BlueZ: 5.69-1
Distribution: arch
Desktop environment: awesome

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2023-09-16:
I have no idea what this is about.

killertofus, 2023-09-16:
if idle for a while the bluetooth disconnects in blueman it "connects" but it doesn't output anything, with other bluetooth applications ie blueberry it doesn't do that

infirit, 2023-09-22:
```
Sep 15 22:46:55 killertofus pulseaudio[2893]: Failed to open module module-bluez5-device.so: module-bluez5-device.so: cannot open shared object file: No such file or directory

```
You have either missing or broken PulseAudio Bluetooth support.

closed 2023-11-30 by github-actions[bot]