blueman-project/blueman #2143: Remote Android --> Raspbian

paolosezart, 2023-09-24:
blueman: 2.1.4
BlueZ: 5.55
Distribution: Raspberry pi4
Desktop environment: Raspbian x64

* [✔ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I am using Blueman on Raspbian OS. It is a very handy programme. I am currently using the Bluetooth PAN and file transfer functions, it all works fine. But I have not been able to connect my android as keyboard and mouse to control Raspbian. That is, the devices connect, but only file transfer and LAN are available to me. On Android there is a great programme "Bluetooth Keyboard & Mouse" https://play.google.com/store/apps/details?id=io.appground.blek&hl=en.
and I use to control my Android TV and occasionally my Windows 10 OS computer.

Can I configure Blueman to work with this programme so that I can control my Raspberry from my phone?

cschramm, 2023-09-25:
I have no idea how that app works, but assuming that it somehow exposes additional services on the Android device, I'd expect the device to advertise HID services, BlueZ should get connected when you connect to the device and your input system should pick them up.

You can see exposed services in the info dialog from the device's right-click menu.

infirit, 2023-09-25:
I wouldn't bother with an app like thst. Raspbian has everything to do remote desktop and remote shell. Look for `xrdp`, `ssh` and `vnc`.

paolosezart, 2023-09-27:
> I wouldn't bother with an app like thst. Raspbian has everything to do remote desktop and remote shell. Look for `xrdp`, `ssh` and `vnc`.

Now I do that, but periodically raspberry pi loses connection with the router. And I have to restart the wifi adapter with the commands:
```
sudo ifconfig wlan1 down
sleep 2
sudo ifconfig vlan1 up
```

And I wanted to be able to control it via bluetooth. And that would be the ideal solution in this case.

infirit, 2023-09-27:
> Now I do that, but periodically raspberry pi loses connection with the router. And I have to restart the wifi adapter with the commands:
> ```
> sudo ifconfig wlan1 down
> sleep 2
> sudo ifconfig vlan1 up
> ```
> 

Maybe something like this will help.

https://feriman.com/automatically-reconnect-to-wifi-on-raspberry-pi/

> And I wanted to be able to control it via bluetooth. And that would be the ideal solution in this case.

If you previously connected the phone to the PI you may need to remove it so BlueZ can discover the new service(s). Do make sure the app is running while adding it.

closed 2023-12-04 by github-actions[bot]