blueman-project/blueman #2158: AttributeError: 'NoneType' object has no attribute 'prepend_search_path'

VPanteleev-S7, 2023-10-08:
blueman: 2.3.5
BlueZ: 5.69
Distribution: Arch Linux
Desktop environment: i3

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Occasionally I see the following crash in my journal:

```
gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 40, in <module>
    app = BluemanApplet()
          ^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/blueman/main/Applet.py", line 19, in __init__
    setup_icon_path()
  File "/usr/lib/python3.11/site-packages/blueman/Functions.py", line 155, in setup_icon_path
    ic.prepend_search_path(ICON_PATH)
    ^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'prepend_search_path'
```

I don't intentionally start `blueman-applet`, but I think `blueman-manager` tries to start it automatically.

Hope this helps.

cschramm, 2023-10-08:
No default icon theme is typically a symptom of no (proper) `$DISPLAY`. Neither `blueman-applet` nor `blueman-manager` can work without one. `blueman-applet` typically gets invoked by D-Bus (via name `org.blueman.Applet`).

Your D-Bus activation environment might lack proper setup. The quick fix is `dbus-update-activation-environment DISPLAY`. A standard setup should work fine (see https://github.com/blueman-project/blueman/issues/1636#issuecomment-980077862). One source of breakage that we identified is users running `startx` with client options, so that it does not use `.xinitrc`. Another one is a broken `.xinitrc`, of course.

VPanteleev-S7, 2023-10-08:
Yes, that would explain it. I masked the `systemctl --user import-environment` call that my distribution normally does.

Perhaps a more explicit error would be in order, but feel free to close this issue. Thanks!

FWIW, the reason why I do the above is that I may run more than one X session at a time (e.g. Xorg and Xwayland under sway, or Xorg on the iGPU + Xorg on the NVIDIA GPU), so having that there caused systemd units to go to some arbitrary / most recently launched X instance, which is often not helpful.

cschramm, 2023-10-08:
Current code actually triggers a notification from blueman-manager when it might have detected that situation.

VPanteleev-S7, 2023-10-08:
I don't think I've seen such a notification, though that might be because I run `blueman-manager` directly, so it has a `DISPLAY`. (I just want the manager for the UI, I don't need the applet.)

closed 2023-12-15 by github-actions[bot]