blueman-project/blueman #2161: Devices window closes when BT adapter is enabled

MartinGebert, 2023-10-09:
blueman: 2.3.5
BlueZ: 5.68
Distribution: Gentoo
Desktop environment: LXQt

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

This is a minor usability issue and for me a tiny but frequent nuisance, and I wonder if it can be solved:

I've configured Bluetooth to start with the adapter disabled; this also applies when it's waking up from hibernating. In case I want to use BT I usually left click the tray icon, and the devices window appears, offering to enable BT in a dialog. So far this is exactly what I want.

When I click the "Enable Bluetooth" button the devices window is closed immediately, while BT has been enabled as expected. This is my issue; I would like to see the available devices **without** having to open the devices window again. OTOH closing the devices window probably makes sense when **disabling** the adapter, and I assume that's the reason it's implemented this way.

This can be worked around by using the context menu of the tray icon (on my system right instead of left button click) to enable the adapter, then open the devices window, but I haven't managed yet to convince my brain to adapt this procedure.

Is there a configuration or other way to customize this behaviour? If not, would you consider it as a feature request, please?

infirit, 2023-10-09:
It's really a bug. For some reason `check_bluetooth_status` thinks enabling/powering failed. The error `Failed to enable bluetooth`

MartinGebert, 2023-10-10:
> For some reason check_bluetooth_status thinks enabling/powering failed.

If this really were an error it should IMHO show an appropriate message, not just close the window and leave the user clueless.

infirit, 2023-10-10:
Tldr, we need to fix this in blueman :-).

It's a timing bug in blueman manager. What happens is that we ask Bluetooth to be powered/enabled but don't wait long enough for it to finish. We then ask too early and are told Bluetooth is not powered/enabled and erroneously close the window.

Sleeping for half a second between powering and asking if it succeeded fixes it. But I like to take the opportunity to change how this is handled in blueman manager. There is actually no reason to close the window as long as we have the (modal) dialog over the main window and prevent the user from interacting with it.

infirit, 2023-10-10:
@MartinGebert if you're able feel free to test #2163.

MartinGebert, 2023-10-12:
@infirit Tried that by replacing `Manager.py`in my `.../sites-packages/blueman/main` dir with the changed file from that PR, and starting `blueman-applet`from the console. Something else seems to changed, too; with BT disabled I now see disabled entries in the tray context menu, and can't open the devices window:

[image]

Enabling BT enables those, but I still can't open the window. I see this errors messages on the console every time I try to open:

```
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/DBusProxies.py", line 52, in call_finish
    proxy.call_finish(resp)
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.UnknownMethod: Keine derartige Schnittstelle »org.freedesktop.Application« des Objekts im Pfad /org/blueman/Manager (19)
```

I will probably have to wait for the complete package to become available, or try to install the current development branch manually.

cschramm, 2023-10-13:
Applying recent changes to 2.3.5 is not a good idea. Gentoo has a blueman-9999.ebuild that builds from git. Testing the Manager.py changes on top of that should work.

closed 2024-02-20 by cschramm