blueman-project/blueman #2170: problem with playing headphones redme bugs 3 pro

saber716rus, 2023-10-15:
blueman:2.3.5+mint1+faye
BlueZ: 5.66-1
Distribution: lmde6
Desktop environment:cinnamon

the problem is manifested by the loss of sound in one of the headphones, by long switching or restarting the PC. The sound is restored.
url hw-probe https://linux-hardware.org/?probe=ea047f0b15

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

serainox420, 2023-10-17:
Do you use pulseaudio and blueman?
If so, make sure pulseaudio-module-bluetooth is installed.

cschramm, 2023-10-24:
Never heard of such an issue. Did you try restarting your sound server or re-connecting the device?

saber716rus, 2023-10-29:
> Never heard of such an issue. Did you try restarting your sound server or re-connecting the device?

if I reboot, the problem disappears for a while, but then it reappears.

saber716rus, 2023-10-29:
> Do you use pulseaudio and blueman? If so, make sure pulseaudio-module-bluetooth is installed.

just installed it, I will monitor the problem.

cschramm, 2023-10-30:
Not sure, but I think LMDE6 uses PipeWire, not pulseaudio, so the package is pointless (also, if it was not installed and pulseaudio would be your sound server, connecting the device as an audio sink would not have worked anyway).

Restarting your sound server should probably be something along `systemctl --user restart wireplumber pipewire pipewire-pulse`, re-connecting the device would be disconnect and connect e.g. in blueman.

closed 2024-01-06 by github-actions[bot]