blueman-project/blueman #2184: Antimicrobial Folding Ergo Keyboard not found

jockep123, 2023-11-07:
blueman:blueman-applet --loglevel debug
blueman-applet 17.31.39
BlueZ: bluetoothd -v  ->   5.64
Distribution:Ubuntu 22.04.3 LTS
Desktop environment:GNOME 42.9

* [ X ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The /var/log/syslog file didn't give any entry lines at all when following https://wiki.ubuntu.com/DebuggingBluetooth

Blueman log:
I just clicked search in the blueman gui
https://pastebin.com/aAEnpj1r

The device I am trying to pair is a Antimicrobial Folding Ergo Keyboard from Targus (https://eu.targus.com/products/antimicrobial-folding-ergo-keyboard-nordic-akf003no).

cschramm, 2023-11-07:
> blueman-applet 17.31.39

There's absolutely nothing like that.

> Blueman log: I just clicked search in the blueman gui https://pastebin.com/aAEnpj1r

There's a 1C:03:2C:59:BF:F6 visible. Is it the desired device? Might be hidden in blueman-manager due to no name. Check the "View" menu.

jockep123, 2023-11-07:
> > blueman-applet 17.31.39
> 
> There's absolutely nothing like that.

Well, that is what the command above echoed in the first line :) 

> 
> > Blueman log: I just clicked search in the blueman gui https://pastebin.com/aAEnpj1r
> 
> There's a 1C:03:2C:59:BF:F6 visible. Is it the desired device? Might be hidden in blueman-manager due to no name. Check the "View" menu.

Hmm... it could be. Why does it not appear in the Bluetooth Devices gui?

infirit, 2023-11-07:
> > There's a 1C:03:2C:59:BF:F6 visible. Is it the desired device? Might be hidden in blueman-manager due to no name. Check the "View" menu.
> 
> Hmm... it could be. Why does it not appear in the Bluetooth Devices gui?

If the keyboard is not sending a device name we hide them in the list. If you look in the View menu and untick the box like @cschramm it should now show up.

jockep123, 2023-11-07:
> > > There's a 1C:03:2C:59:BF:F6 visible. Is it the desired device? Might be hidden in blueman-manager due to no name. Check the "View" menu.
> > 
> > 
> > Hmm... it could be. Why does it not appear in the Bluetooth Devices gui?
> 
> If the keyboard is not sending a device name we hide them in the list. If you look in the View menu and untick the box like @cschramm suggested it should now show up.

Yes, now it works. Thanks! :)

cschramm, 2023-11-08:
For reference: This should not happen with #1994 which will be in 2.4.0.

closed 2024-01-15 by github-actions[bot]