blueman-project/blueman #2185: Directory for transferred files can't be configured (files appear in ~/.cache/obexd)

rmwiseman, 2023-11-08:
blueman: 2.3.5 (at least, that's the version reported by the manager's help menu; dpkg reports 2.3.5+mint1+vera)
BlueZ: 5.64-0ubuntu1 (via dpkg)
Distribution: Linux Mint 21.2
Desktop environment: Cinnamon (verson: 5.8.4+victoria)

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I've looked in the Preferences and in /etc/bluetooth/* but can't find anywhere to specify where I want bluetooth-transferred files to be put.  I vaguely recall configuring this (to my Desktop, I believe) in an earlier version or possibly when Mint used a different bluetooth manager.  But now, I don't seem to be able to configure it, and the files are put in ~/.cache/obedx which is not easy for me to remember!

They get transferred fine, it's just that they're not put somewhere useful for me.

Looking at the debug output of blueman-applet (as per the troubleshooting guide) this is confirmed by one of the lines:

`blueman-applet 09.13.32 DEBUG    Base:60 do_g_properties_changed: /org/bluez/obex/server/session2 {'Source': 'E8:B1:FC:55:E5:AA', 'Destination': 'AC:67:5D:C7:DD:81', 'Root': '/home/richard/.cache/obexd'}`

but it doesn't say elsewhere where that is configured or if it's a fallback location.

Is it possible to configure/reset where transferred files appear?

infirit, 2023-11-08:
Can you post a whole debug log so we can see what happens before and after the transfer.

rmwiseman, 2023-11-08:
Thanks for the reply.

Here's my debug output: https://gist.github.com/rmwiseman/b0100609728ec7977c8659044e016e14
(I added the blank lines before transferring the file.)

Before I did any of this, I changed the Incoming Folder (as per your guidance) to my Desktop, but the file still appeared in the same place.

infirit, 2023-11-12:
The whole transfer is missing in the log after it was accepted. This is how it should look like, https://gist.github.com/infirit/6ba615adf1d5a4fa44f71b4f55beefa9.

Is gnome-bluetooth running? Both will fight for the transfer and it never ends well.

rmwiseman, 2023-11-13:
Thanks @infirit, that's interesting and confusing!

> Is gnome-bluetooth running? Both will fight for the transfer and it never ends well.

```
$ sudo ps -ef | grep -i bluetooth
root         936       1  0 07:53 ?        00:00:00 /usr/lib/bluetooth/bluetoothd
richard     3948    2404  0 07:54 ?        00:00:00 /usr/lib/bluetooth/obexd
```

(Obviously my `grep` command is returned above too, but that's not relevant here.)

infirit, 2023-11-13:
Ok, the Unicode error may be relevant. We changed how we construct our bluez classes in what will be one 2.4 which should handle this error. Can you try a git build from https://blueman.cschramm.eu/debian/ ?

These should work on mint.

rmwiseman, 2023-11-14:
Thanks @infirit, I don't know whether I did it right: I just installed via apt, but the new version number mentioned git so I assumed it was the right one!  However, it _seems_ to have worked.  At least, the file now appears on my Desktop, the debug output includes the line:

```
blueman-applet 14.46.03 DEBUG    Transfer:46 _properties_changed: /org/bluez/obex/server/session1/transfer0 Status complete
```

and there's now no Unicode error.

I trust those are all good signs!

If so, when would be the right time to remove the git build from my apt lists?

infirit, 2023-11-14:
Yes that is how it should be. This change will be in version 2.4.

rmwiseman, 2023-11-15:
Thanks again @infirit, I really appreciate your time and assistance.

I didn't notice a notification, but I'll keep an eye out next time I do a transfer.

closed 2023-11-15 by rmwiseman