blueman-project/blueman #2187: Local Services enigma

birdie-github, 2023-11-12:
I'm lost in this dialog.

What's NAP? What is it used for?

Likewise, what about PAN and DUN?

Can we have some descriptive text or `?` around all the options in this dialog?

Thank you.

cschramm, 2023-11-12:
What language do you refer to? The original message as well as the translations that I just checked write out NAP.

PAN and DUN are not described, yes. You should get an idea when looking at the related plugins, but of course there's no obvious connection from the services dialog to the plugins dialog.

To be honest, I'm not sure if it's helpful to users that do not understand those terms and most probably do not want to use the settings anyway, that PAN is for, well, personal networks and DUN for, well, dial-up networks. In the end PAN and DUN are profiles, named exactly like that, provided by devices and settings for them only make sense when the user already knows that and what he wants to use.

infirit, 2023-11-12:
> PAN and DUN are not described, yes.

I added tooltips, I can put them under a ? icon if that makes it more discoverable.

closed 2024-01-19 by github-actions[bot]