blueman-project/blueman #2190: BM30X Bluetooth Mouse reconnection issues

Syewm, 2023-11-12:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2023-11-12:
You're saying when you pair via bluetoothctl the device stays connected but when you pair via blueman it does not? That's very surprising.

Does bluetoothd give you any logs on the disconnects?

Syewm, 2023-11-12:
When i pair through the Blueman it fails to reconnect when the device goes to sleep. If i do the same thing through terminal using bluetoothctl it always reconnects when the device comes back from sleep.

When i type bluetoothd in terminal it gives me:

"D-Bus setup failed: Connection ":1.134" is not allowed to own the service "org.bluez" due to security policies in the configuration file"

cschramm, 2024-01-15:
bluetoothd logs should be browsable with `sudo journalctl -u bluetooth.service`.

closed 2024-01-23 by github-actions[bot]