blueman-project/blueman #2197: `--loglevel`'s argument isn't checked

kuraga, 2023-11-16:
blueman: 2.3.5
BlueZ: 5.68
Distribution: Calculate Linux (Gentoo-based)
Desktop environment: XFCE 4.18

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

```bash
blueman-applet --loglevel something-incorrect
```
doesn't provide an error.

infirit, 2023-11-16:
It's checked but if an invalid level is provided we default to warning. Why do you want it to error?

kuraga, 2023-11-16:
```bash
blueman-applet --loglevel 0
```
```
(silent)
```

Question: So, there are no messages I need to know?
Answer I see: Yes.
Real answer: No.

infirit, 2023-11-16:
The warning level only shows actual errors and we have few warnings. With normal use there is nothing a regular user should see here.

Is there something you are missing and would like to see?

kuraga, 2023-11-16:
No, thanks.

closed 2023-11-16 by kuraga