blueman-project/blueman #2205: Keyboard Logitech K380

Simonejkk, 2023-12-03:
I can't connect my keyboard at Linux bluetooth. Thanks.

maccaberry, 2023-12-03:
I have the same problem. Just installed Mint xfce and trying to connect Logitech K380 keyboard also. My Logitech M385 mouse connected instantly. But the keyboard doesn't appear in BT search. Just the mouse.

infirit, 2023-12-04:
We hide devices with no name by default so check the view menu to disable this.

For the next release blueman will never hide keyboards to avoid this.

Gojko-M, 2023-12-08:
I have the same problem. I am using `Logitech K380` keyboard for a couple of months now, and each time when I turned laptop on I have to manually connect keyboard via Bluetooth by entering _paring mode_ on keyboard and then connect keyboard from _**Bluetooth Device Manager**_. 

In case laptop goes in sleep mode, I log out or lock, the above process needs to be repeated. 

But from today, when I go through the same process of connecting keyboard,  keyboard connects, but it just does not work, it does not responds to any keystroke, like its not connected at all, and it gets disconnected after couple of minutes. 

```
Distribution:
NAME="Linux Mint"
VERSION="21.2 (Victoria)"

About blueman
blueman-applet 2.3.5
```

Please let me know if any additional information is needed.

shamanInNeed, 2023-12-20:
I have the same problem. I have a Logitech K380 keyboard. Linux Mint cannot find that Device.

pieas-asif, 2023-12-20:
I have the same issue. I have a `Logitech K380` and I cannot find the device in the Linux Mint to Pair.

closed 2024-02-26 by github-actions[bot]