blueman-project/blueman #2212: Add gnome-bluetooth support

kacperpaczos, 2023-12-07:
blueman: 2.3.5
BlueZ:
Distribution: Linux Mint 
Desktop environment: Cinnamon

IPods on Gnome Bluettoth which use: https://gitlab.gnome.org/GNOME/gnome-bluetooth work better than blueman.

Consider add alternative backend.

cschramm, 2023-12-07:
Not an option, but please elaborate on "work better".

labbatt50, 2023-12-25:
When I Open it ***everything*** is greyed out.
Restart to conect, same issue.
Only option is Help and Quit.

[image]

SzymonPaczos, 2023-12-25:
Hello everyone

Kacper Paczos has observed an issue with pairing a laptop running Linux Mint Cinnamon with Apple Air Pods Pro 2. 
While both devices can be successfully paired, they automatically disconnect shortly afterward. Despite the pairing, establishing a connection is not possible. 

Please suggest troubleshooting steps to help identify the root cause of the issue.

cschramm, 2024-01-01:
Audio connections are handled by the system's audio server, usually pulseaudio or PipeWire. Its logs (and maybe bluetoothd's) might already give a hint.

cschramm, 2024-01-02:
pulseaudio logs should be `journalctl --user -u pulseaudio.service` and BlueZ' `sudo journalctl -u bluetooth.service` on your system.

cschramm, 2024-01-02:
I did not get that you're the off-topic one, probably not on Linux Mint then...

I'm not really sure what's going in this issue thread, but it seems like @SzymonPaczos reported an issue with audio connections on behalf of @kacperpaczos who opened the issue with no information on the actual problem and @labbatt50 jumped in with a completely different issue and now tries to follow my hints for @SzymonPaczos / @kacperpaczos.

@labbatt50, please open a new issue if you want us to look into that. I don't understand why you ended up in this thread.

labbatt50, 2024-01-02:
Christopher,

No, Mint was ages ago. PCLinuxOS Xfce now.
I will login later and try to find the correct thread.

Cheers and Happy New Year
L

Get Outlook<https://aka.ms/o0ukef>

closed 2024-03-10 by github-actions[bot]