blueman-project/blueman #2216: stuttery sound when connecting Samsung smartphones via bluetooth to transmit sound on a laptop via smartphones

siwy666, 2023-12-09:
blueman:2.3.5
BlueZ:5.70
Distribution:archcraft
Desktop environment:openbox

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2023-12-12:
Might be an issue of your sound server (typically pulseaudio or PipeWire). I'd try a different one to see if that's the case.

siwy666, 2024-01-17:
When i change  in samsung galaxy a 12 in developer options ( i turn off hd audio, i set coded audio bluetooth to SBC and everything change to default  in sound bluetooth) everything works prefectly.

closed 2024-01-17 by siwy666

infirit, 2024-01-17:
Just speculating but..

The SBC codec bandwith requirement is about 320kb/s max. Compare that to something like LDAC that maxes out at 900kb/s. Aptx is more reasonable depending on the the version which is between 380 and 580kb/s.

If switching to SBC fixes the stuttering it may be because of the higher bandwith requirements. However this shouldn't be a problem with modern hardware and Bluetooth versions. Unless there is radio interference from WiFi, distance or walls.

siwy666, 2024-01-17:
When I turn off HD Sound in Developer options ant other bluetooth settings i set in default sound from Samsung A12 is perfect.