blueman-project/blueman #2217: blueman-applet closes itself immediately

cristianc, 2023-12-10:
blueman: 2.3.5-r2
BlueZ: 5.66-r5
Distribution: Alpine Linux v3.19
Desktop environment: LXQt 1.3.0-r0

When I try to open blueman, I get the following error messages:
`blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Failed to execute program org.blueman.Mechanism: No such file or directory`

Complete blueman-applet output messages can be found at: https://pastebin.com/JynMBrhC

Waiting for a reply,

Thanks.

cschramm, 2023-12-12:
Looks like at least the package's dependencies are incomplete. Your system lacks the libnm GI bindings.

Is `/usr/share/dbus-1/system-services/org.blueman.Mechanism.service` present? Does its `Exec` point to `/usr/libexec/blueman-mechanism`? Does `/usr/libexec/blueman-mechanism` exist and start?

cristianc, 2023-12-12:
> Looks like at least the package's dependencies are incomplete. Your system lacks the libnm GI bindings.

I already tried to install network manager in past but  `blueman.main.DBusProxies.DBusProxyFailed: Error calling StartServiceByName for org.blueman.Mechanism: Failed to execute program org.blueman.Mechanism: No such file or directory` still was shown. Btw, now I've installed `libnm` and I still get the same error messages, as shown by the new output: https://pastebin.com/mkzq2riM

> Is /usr/share/dbus-1/system-services/org.blueman.Mechanism.service present?

Yes.

> Does its Exec point to /usr/libexec/blueman-mechanism?

Yes

> Does /usr/libexec/blueman-mechanism exist and start?

It exists but it wasn't trivial checking if this python script starts. I've run `grep "blueman-mechanism" /var/log/messages` and it returns just the message about grep command. I've run `ps aux | grep "blueman-mechanism"` and it returns just  `2974 alpine    0:00 grep blueman-mechanism`.
I don't know if this script starts, definitively, but I've found a related issue here in this github tracker about that: https://github.com/blueman-project/blueman/issues/1403

cschramm, 2023-12-12:
I meant are you able to run `/usr/libexec/blueman-mechanism` yourself?

cristianc, 2023-12-12:
> I meant are you able to run `/usr/libexec/blueman-mechanism` yourself?

If I run `/usr/libexec/blueman-mechanism`, it just hangs and after some time it returns command prompt. If I run the same command using root privileges, the string `Authorization required, but no authorization protocol specified` is printed twice, the command hangs and finally after some time it returns command prompt.

cschramm, 2023-12-13:
So running it does work and the issue lies in DBus activation somehow.

Just to double check: `/usr/libexec/blueman-mechanism` has an absolute path to python in its shebang, correct?

Does `/usr/libexec/dbus-daemon-launch-helper org.blueman.Mechanism` give anything?

What it should do is check `/usr/share/dbus-1/system-services/org.blueman.Mechanism.service` for the D-Bus service definition, try to start its `SystemdService` - which will fail because there is no systemd, correct? - and eventually run its `Exec` as root - which is exactly what you successfully tested.

cristianc, 2023-12-14:
> So running it does work and the issue lies in DBus activation somehow.

I mean that running `/usr/libexec/blueman-mechanism` doesn't fix the problem: blueman-applet still closes itself immediately in the same way.

> Just to double check: /usr/libexec/blueman-mechanism has an absolute path to python in its shebang, correct?

Opening `/usr/libexec/blueman-mechanism` script with a text editor shows the following shebang: `#!/usr/bin/python` (so, it's an absolute path)

`Does /usr/libexec/dbus-daemon-launch-helper org.blueman.Mechanism give anything?`
I can't run this command because `dbus-daemon-launch-helper` doesn't exist inside `/usr/libexec` directory: `/bin/sh: /usr/libexec/dbus-daemon-launch-helper: not found`

cschramm, 2023-12-14:
Well, that sounds like it could cause the trouble. However, the `blueman` package depends on the `dbus` package and that *does* include `/usr/libexec/dbus-daemon-launch-helper`. It's available to me when I quickly test with `apk add blueman`. Looking at `dbus`'s APKBUILD, the helper is part of a subpackage called `dbus-daemon-launch-helper` and that's where my Alpine knowledge ends. I have no idea how to control that subpackage and why it would not be installed for you, but I reckon that it is required for D-Bus activation to work and is thus a requirement for blueman.

Jakko3, 2024-01-02:
Possibly enabling the OpenRC services is missing. After installing the packages (dbus, dbus-x11, bluez, bluez-obexd, blueman), you also need to enable the services (dbus, bluetooth):
- `doas rc-update add dbus`
- `doas rc-update add bluetooth`

And then reboot. They can also be started without rebooting by `doas rc-service dbus start` and `doas rc-service bluetooth start`. The state can be checked by `rc-service dbus status` and `rc-service bluetooth status`.

Maybe I missed to list some packages to install or services to add. However, basically Blueman in Alpine Linux should work.

cristianc, 2024-01-04:
Hi,

unfortunately, I already added and enabled dbus and bluetooth services, unsuccessfully. This issue was related to dbus broken launch helper.

Jakko3, 2024-01-04:
In postmarketOS (based on Alpine Linux) in Xfce4 & MATE Blueman basically works. I haven't tested LXQt, though.

I still think it's more a question of what additional packages need to be installed and what services are to be enabled (related to Alpine Linux). Let's try to find that out together.

On [pkgs.alpinelinux.org](https://pkgs.alpinelinux.org) -> Page "Contents" -> searching for `*dbus-daemon-launch-helper` (and branch v3.19 and architecture x86) it shows that this command is part of package "dbus-daemon-launch-helper". Can you install that package and test again? (And in case it doesn't work, post the error output, as it may has changed.)

Jakko3, 2024-01-04:
I installed Blueman on postmarketOS LXQt. It works there.

cristianc, 2024-01-07:
> On [pkgs.alpinelinux.org](https://pkgs.alpinelinux.org/) -> Page "Contents" -> searching for *dbus-daemon-launch-helper (and branch v3.19 and architecture x86) it shows that this command is part of [...]

Unfortunately, dbus-daemon-launch-helper package (https://pkgs.alpinelinux.org/contents?branch=v3.19&name=dbus%2ddaemon%2dlaunch%2dhelper&arch=x86&repo=main) was already installed when I upgraded from 3.18 to 3.19. Now, I've installed a fresh 3.19 alpine installation and `/usr/libexec/dbus-daemon-launch-helper` is in place, instead (now, blueman it's showing up and working).

closed 2024-01-07 by cristianc