blueman-project/blueman #2219: Gamepad unusable after blueman pairing, possibly because blueman does not set up bonding with the device, works fine with bluetoothctl

rs-gh-asdf, 2023-12-11:
blueman: 2.3.5-3
BlueZ: 5.68-0ubuntu1.1
Distribution: Xubuntu 23.10
Desktop environment: Xfce 4.18

* [ ✅ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Other bug reports for reference:
[Launchpad](https://bugs.launchpad.net/ubuntu/+source/blueman/+bug/2046084)
[Bluez Github](https://github.com/bluez/bluez/issues/673)

After an update to bluez that was done to patch [CVE-2023-45866](https://nvd.nist.gov/vuln/detail/CVE-2023-45866), I was still able to pair/connect with my bluetooth gamepad from blueman, but it was not recognized by any software on my computer, not even jstest or jstest-gtk. I think that this is because no `/dev/input/js0` was created. Downgrading bluez to a version without the security patch resolved the issue, but required running an insecure version of bluetoothd. This security update affected how bluetoothd handles HID encryption by default, so it makes some sense that HIDs would break.

However, the same bluetooth device paired on the same version of bluetoothd but with `bluetoothctl` instead of blueman had no issues. For this reason I believe that the issue actually exists in how blueman handles pairing for HIDs, specifically gamepads. I think that blueman needs to use `src/adapter.c:adapter_bonding_attempt()` and `src/device.c:device_set_bonded()`, but I am just an armchair developer in this situation.

I know that the prompt said to use an external host to paste my logs. I do not think that these logs are too long and having them all on the same page I think is helpful for easy comparison. If I had very long logs to paste I would put them in an external host and if someone has an issue with my logs being directly in this bug report then I will move them, but I think this is helpful.

blueman-applet debug logs which produce a non-working connection:

```
blueman-applet 16.31.25 DEBUG    Manager:52 _on_object_added: Device1: /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF
blueman-applet 16.31.25 INFO     Applet:95 on_device_created: Device created /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF
blueman-applet 16.31.25 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF {'Address': 'E4:17:DE:AD:BE:EF', 'AddressType': 'public', 'Name': '8BitDo Pro 2', 'Alias': '8BitDo Pro 2', 'Class': 9480, 'Icon': 'input-gaming', 'Paired': False, 'Bonded': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'RSSI': -42, 'Connected': False, 'UUIDs': ['00000011-0000-1000-8000-redacted', '00000100-0000-1000-8000-redacted', '00001002-0000-1000-8000-redacted', '00001124-0000-1000-8000-redacted', '00001200-0000-1000-8000-redacted', '00000001-0000-1000-8000-redacted'], 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
blueman-applet 16.31.28 INFO     RecentConns:112 notify    : <Device.Device object at 0x7f398df3ffc0 (blueman+bluez+Device+Device at 0x2208760)> 00000000-0000-0000-0000-000000000000
blueman-applet 16.31.29 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 16.31.29 INFO     GameControllerWakelock:72 xdg_screensaver: Number of locks: 1
blueman-applet 16.31.29 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF {'Connected': True}
blueman-applet 16.31.29 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF {'Modalias': 'usb:v045Ep02E0d0903'}
blueman-applet 16.31.29 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF {'UUIDs': ['00001124-0000-1000-8000-redacted', '00001200-0000-1000-8000-redacted'], 'ServicesResolved': True, 'WakeAllowed': True}
blueman-applet 16.31.30 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Discovering': False}
blueman-applet 16.31.30 DEBUG    DiscvManager:92 on_adapter_property_changed: prop Discovering False
blueman-applet 16.31.30 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF {'Paired': True}
```

bluetoothd debug logs when pairing with blueman:

```
Dec 10 16:35:35 laptop bluetoothd[34128]: src/device.c:device_create() dst E4:17:DE:AD:BE:EF
Dec 10 16:35:35 laptop bluetoothd[34128]: src/device.c:device_new() address E4:17:DE:AD:BE:EF
Dec 10 16:35:35 laptop bluetoothd[34128]: src/device.c:device_new() Creating device /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF
Dec 10 16:35:35 laptop bluetoothd[34128]: src/device.c:device_set_legacy() legacy 0
Dec 10 16:35:35 laptop bluetoothd[34128]: src/device.c:device_set_rssi_with_delta() rssi -43
Dec 10 16:35:35 laptop bluetoothd[34128]: src/device.c:device_set_class() /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF 0x002508
Dec 10 16:35:37 laptop bluetoothd[34128]: src/device.c:connect_profiles() /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF (all), client :1.5140
Dec 10 16:35:37 laptop bluetoothd[34128]: src/device.c:btd_device_set_temporary() temporary 0
Dec 10 16:35:37 laptop bluetoothd[34128]: src/device.c:connect_profiles() Resolving services for /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF
Dec 10 16:35:37 laptop bluetoothd[34128]: src/shared/mgmt.c:send_request() [0x0000] command 0x0033
Dec 10 16:35:37 laptop bluetoothd[34128]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x002a
Dec 10 16:35:37 laptop bluetoothd[34128]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x0033 complete: 0x00
Dec 10 16:35:37 laptop bluetoothd[34128]: src/adapter.c:add_accept_list_complete() E4:17:DE:AD:BE:EF added to kernel accept list
Dec 10 16:35:37 laptop bluetoothd[34128]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0013
Dec 10 16:35:37 laptop bluetoothd[34128]: src/adapter.c:discovering_callback() hci0 type 7 discovering 0 method 0
Dec 10 16:35:37 laptop bluetoothd[34128]: src/adapter.c:trigger_start_discovery()
Dec 10 16:35:37 laptop bluetoothd[34128]: src/adapter.c:cancel_passive_scanning()
Dec 10 16:35:37 laptop bluetoothd[34128]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x000b
Dec 10 16:35:37 laptop bluetoothd[34128]: src/adapter.c:connected_callback() hci0 device E4:17:DE:AD:BE:EF connected eir_len 19
Dec 10 16:35:38 laptop bluetoothd[34128]: src/device.c:device_probe_profiles() Probing profiles for device E4:17:DE:AD:BE:EF
Dec 10 16:35:38 laptop bluetoothd[34128]: profiles/input/device.c:input_device_register() /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF
Dec 10 16:35:38 laptop bluetoothd[34128]: src/service.c:btd_service_ref() 0x55950c569870: ref=2
Dec 10 16:35:38 laptop bluetoothd[34128]: src/service.c:change_state() 0x55950c569870: device E4:17:DE:AD:BE:EF profile input-hid state changed: unavailable -> disconnected (0)
Dec 10 16:35:38 laptop bluetoothd[34128]: src/device.c:device_svc_resolved() /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF err 0
Dec 10 16:35:38 laptop bluetoothd[34128]: src/device.c:connect_profiles() /org/bluez/hci0/dev_E4_17_DE_AD_BE_EF (all), client :1.5140
Dec 10 16:35:38 laptop bluetoothd[34128]: profiles/input/device.c:input_device_connect()
Dec 10 16:35:38 laptop bluetoothd[34128]: profiles/input/device.c:ioctl_is_connected() Can't get HIDP connection info
Dec 10 16:35:38 laptop bluetoothd[34128]: src/service.c:change_state() 0x55950c569870: device E4:17:DE:AD:BE:EF profile input-hid state changed: disconnected -> connecting (0)
Dec 10 16:35:38 laptop bluetoothd[34128]: src/shared/mgmt.c:send_request() [0x0000] command 0x0050
Dec 10 16:35:38 laptop bluetoothd[34128]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x0050 complete: 0x00
Dec 10 16:35:39 laptop bluetoothd[34128]: src/shared/mgmt.c:can_read_data() [0x0000] event 0x0009
Dec 10 16:35:39 laptop bluetoothd[34128]: src/adapter.c:new_link_key_callback() hci0 new key for E4:17:DE:AD:BE:EF type 4 pin_len 0 store_hint 0
Dec 10 16:35:39 laptop bluetoothd[34128]: src/device.c:device_bonding_complete() bonding (nil) status 0x00
[... 31 lines cut ...]
Dec 10 16:45:23 laptop bluetoothd[34128]: src/agent.c:agent_unref() 0x55950c553a10: ref=1
Dec 10 16:45:23 laptop bluetoothd[34128]: src/adapter.c:resume_discovery()
Dec 10 16:45:23 laptop bluetoothd[34128]: src/shared/mgmt.c:can_read_data() [0x0000] command 0x0019 complete: 0x00
Dec 10 16:45:23 laptop bluetoothd[34128]: src/adapter.c:pair_device_complete() Success (0x00)
Dec 10 16:45:23 laptop bluetoothd[34128]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr E4:17:DE:AD:BE:EF type 0 status 0x0
Dec 10 16:45:23 laptop bluetoothd[34128]: src/device.c:device_bonding_complete() bonding (nil) status 0x00
[... cut ...]
Dec 10 16:45:25 laptop bluetoothd[34128]: src/service.c:change_state() 0x55950c5678f0: device E4:17:DE:AD:BE:EF profile input-hid state changed: connecting -> connected (0)
Dec 10 16:45:25 laptop bluetoothd[34128]: src/device.c:device_profile_connected() input-hid Success (0)
Dec 10 16:45:25 laptop bluetoothd[34128]: src/device.c:device_profile_connected() returning response to :1.5668
Dec 10 16:45:25 laptop bluetoothd[34128]: src/service.c:btd_service_ref() 0x55950c5678f0: ref=3
Dec 10 16:45:25 laptop bluetoothd[34128]: plugins/policy.c:service_cb() Added input-hid reconnect 0
Dec 10 16:45:25 laptop bluetoothd[34128]: profiles/input/device.c:encrypt_notify()
```

rs-gh-asdf, 2023-12-11:
As a test, I downloaded and compiled the latest commit of blueman and tried with that, but I still had the same issue of my gamepad not being usable after being paired. I'm not sure if I missed a step in the compilation, though, I did:

```
git clone https://github.com/blueman-project/blueman.git
cd blueman
./autogen.sh
make
sudo make install
export PYTHONPATH=/usr/local/lib/python3.11/site-packages
dbus-update-activation-environment
blueman-applet
```

And then in a second terminal I did:
```
export PYTHONPATH=/usr/local/lib/python3.11/site-packages
dbus-update-activation-environment
blueman-manager
```

infirit, 2023-12-11:
Can you provide the exact steps in blueman and bluetoothctl? Specifically interested whether you used Pair and then Connect or just Connect.

cschramm, 2023-12-11:
From the logs it looks like an unpaired connection attempt from blueman but a dedicated pairing attempt from bluetoothctl.

rs-gh-asdf, 2023-12-11:
@infirit In bluetoothctl I do:

```
scan on
// wait for the device to show up
pair MAC_ADDRESS
connect MAC_ADDRESS
trust MAC_ADDRESS
 ```

in blueman I previously just clicked on "connect" since that is what I have done in the past using blueman and the same controller. Just now I did:

- search
- right-click "pair" (automatically connects and then disconnects)
- right-click "connect"

And that seems to have worked. My gamepad now shows up in jstest and in different software. Trusting in blueman after these steps allows the gamepad to automatically reconnect when it is turned on.

Please excuse my ignorance earlier with baseless speculation - I don't actually know how bluetooth works. I had always just used "connect" in the past, it is the first option in the right-click menu, and it is what happens when you double-click on the device. That worked in the past before the latest bluez update. I suppose that this is just user error on my part.

cschramm, 2023-12-12:
I'm not actually sure. It rather seems like a BlueZ bug, but it could be tricky. When you connect to the unpaired device, pairing gets initiated remotely. You see in the logs that it completes successfully. Then the HIDP connection gets rejected because the device is not marked as bonded (yet). I think it eventually should be and that might be some race condition. Does e.g. `bluetoothctl info` show it as bonded afterward? What I do not get is why HID is considered connected even though the connection got refused.

You could try https://packages.ubuntu.com/noble/bluez to see if the problem sticks with recent versions of BlueZ.

cschramm, 2023-12-13:
So the device does not get marked as bonded at any time if it's not explicitly paired. That's unexpected to me and it's also unexpected that it leads to seemingly successful but useless connections. It clearly seems like some BlueZ issue to me.

rs-gh-asdf, 2023-12-13:
Okay, thanks for all your help. I'll move this back to the bluez github issue.

closed 2023-12-13 by rs-gh-asdf

Aleksanaa, 2024-12-16:
There are more discussions about the bug happening in said bluez issue page. Could you review it again and confirm if this is a security issue that needs to be addressed on blueman's part?

cschramm, 2024-12-16:
Not sure if I'm missing something there, but what I read is that blueman and bluetoothctl are in line regarding connecting to an unpaired device.

There are people reporting that using bluetoothctl to pair the device would be a solution. That is correct, but they are missing that the issue is with connecting to the unpaired device, not with explicit pairing. blueman also provides a "Pair" action that does not cause the issue. bluetoothctl also provides a `connect` action that does cause the issue if the device is not paired yet.

There is also one report of solving the issue by switching to bluedevil. If I remember correctly, bluedevil always pairs new devices explicitly in some kind of wizard, so this is nothing else than explicit pairing with one of the other clients.

I'm not 100 % sure about https://github.com/bluez/bluez/issues/673#issuecomment-2525249151. I assume that's where you see a potential security issue. It sounds like the controller knows the computer as paired and automatically connects, probably leading to a similar situation, but I do not get how blueman (probably its agent) would be involved there. Could be that the whole sequence cannot happen if there is no agent. That's not blueman specific then either, though.