blueman-project/blueman #2221: dualshock 3 asking for pin code after update (doesn't work)

le-dude-520, 2023-12-13:
blueman:
BlueZ:
Distribution: Linux Mint 21.2
Desktop environment: Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

le-dude-520, 2023-12-13:
not a big techno wiz so I'll just say my ps3 controller paired seamlessly before the last update, now however it asks for a pin code that doesn't work (I tried most of the ones I googled like 0000)

cschramm, 2023-12-14:
That's https://github.com/bluez/bluez/issues/673. Adding an `/etc/bluetooth/input.conf` with `ClassicBondedOnly=false` should fix it. For some reason, `ClassicBondedOnly=true`, which is the new default, breaks pairing for some devices if it's not initiated explicitly ... and as described in https://github.com/bluez/bluez/issues/673#issuecomment-1854695143 you cannot explicitly pair that controller.

le-dude-520, 2023-12-14:
Hey thanks for your input, I tried what you suggested, but it still asks for a pin code.. Can I downgrade to the previous build?

infirit, 2023-12-14:
The file should contain the following and it should from then on work like before. This does open you up for keystroke injection from CVE-2023-45866.

```
[General]
ClassicBondedOnly=false
```

closed 2023-12-14 by infirit