blueman-project/blueman #2222: Blueman's GUI goes off if i down the non-default adapter

Flashwalker, 2023-12-14:
blueman: 2.2.4-1
BlueZ: 5.64-0ubuntu1.1
Distribution: Ubuntu 22.04
Desktop environment: xfce4.18

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Let's say i have two adapters
**Intel** - integrated - hci1
**Realtek** - USB dongle - hci0

And i set as default the **Realtek** hci0

Now If i DOWN the integrated non-default one:
```sh
sudo hciconfig hci1 down
```
Blueman's GUI goes off, even if USB dongle adapter is UP and set as default

cschramm, 2023-12-14:
By GUI you mean blueman-manager? Please run it as `blueman-manager --loglevel debug` from a terminal and provide the output of when you power down the adapter.

infirit, 2023-12-14:
I can't replicate this with my usb dongles on main.

Flashwalker, 2023-12-15:
> By GUI you mean blueman-manager?

Yes

**adapters enumeration:**
```sh
$ hciconfig -a
hci1:	Type: Primary  Bus: USB
	BD Address: 8C:87:0B:00:08:A8  ACL MTU: 1021:6  SCO MTU: 255:12
	UP RUNNING PSCAN 
	RX bytes:2013 acl:0 sco:0 events:198 errors:0
	TX bytes:35669 acl:0 sco:0 commands:198 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'rtl8761bw'
	Class: 0x7c010c
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Laptop
	HCI Version: 5.1 (0xa)  Revision: 0xdfc6
	LMP Version: 5.1 (0xa)  Subversion: 0xd922
	Manufacturer: Realtek Semiconductor Corporation (93)

hci0:	Type: Primary  Bus: USB
	BD Address: 7C:67:A2:EC:50:17  ACL MTU: 1021:5  SCO MTU: 96:6
	UP RUNNING PSCAN 
	RX bytes:39587025 acl:170 sco:0 events:4458264 errors:0
	TX bytes:2143624279 acl:4081383 sco:0 commands:98047 errors:0
	Features: 0xff 0xfe 0x0f 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'intel'
	Class: 0x7c010c
	Service Classes: Rendering, Capturing, Object Transfer, Audio, Telephony
	Device Class: Computer, Laptop
	HCI Version: 4.2 (0x8)  Revision: 0x1000
	LMP Version: 4.2 (0x8)  Subversion: 0x1000
	Manufacturer: Intel Corp. (2)
```
 
**blueman-manager start:**
```sh
$ blueman-manager --loglevel debug
blueman-manager 03.37.04 INFO     PluginManager:85 load_plugin: ['PulseAudioProfile', 'Info', 'Services', 'Notes']
blueman-manager 03.37.04 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'>
blueman-manager 03.37.04 INFO     PulseAudioUtils:193 pa_context_event: 1
blueman-manager 03.37.04 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.manager.Info.Info'>
blueman-manager 03.37.04 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.manager.Services.Services'>
blueman-manager 03.37.04 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.manager.Notes.Notes'>
blueman-manager 03.37.04 INFO     PulseAudioUtils:193 pa_context_event: 2
blueman-manager 03.37.04 INFO     Manager:121 on_dbus_name_appeared: org.bluez :1.29
blueman-manager 03.37.04 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '7C:67:A2:EC:50:17', 'AddressType': 'public', 'Name': 'smarttop', 'Alias': 'intel', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-manager 03.37.04 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci1 {'Address': '8C:87:0B:00:08:A8', 'AddressType': 'public', 'Name': 'smarttop #2', 'Alias': 'rtl8761bw', 'Class': 8126732, 'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 180, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-manager 03.37.04 DEBUG    DeviceList:190 set_adapter: Setting adapter to: None
blueman-manager 03.37.04 DEBUG    ManagerToolbar:69 on_adapter_changed: toolbar adapter /org/bluez/hci0
blueman-manager 03.37.04 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_F5_32_D7_25_37_10 {'Address': 'F5:32:D7:25:37:10', 'AddressType': 'public', 'Name': 'FreeMotion B535', 'Alias': 'FreeMotion B535', 'Class': 2360324, 'Icon': 'audio-headset', 'Paired': True, 'Trusted': True, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': ['0000110b-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb'], 'Modalias': 'bluetooth:v05D6p000Ad0240', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
[... cut ...]
    super().device_remove_event(device)
  File "/usr/lib/python3/dist-packages/blueman/gui/DeviceList.py", line 173, in device_remove_event
    assert tree_iter is not None
AssertionError
...
```

**at this point i do:**
```sh
$ sudo hciconfig hci0 down
```

And blueman-manager becomes disabled and i can't open the devices list anymore
```sh
...
blueman-manager 03.38.06 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Class': 0, 'Powered': False, 'Discovering': False}
```

[image]

[image]

infirit, 2023-12-16:
Can't replicate on 2.3.5 either.

cschramm, 2024-01-01:
Sounds like something from the past then. I recommend trying https://packages.ubuntu.com/lunar/blueman.

Flashwalker, 2024-01-06:
> Sounds like something from the past then. I recommend trying https://packages.ubuntu.com/lunar/blueman.

Lunar's version depends on `python3` >= 3.11
but Jammy have only 3.10

cschramm, 2024-01-06:
Ah, sorry, missed that. You could try the blueman-git package from https://launchpad.net/~pipewire-debian/+archive/ubuntu/pipewire-upstream for 2.3.5 or my snapshots on https://blueman.cschramm.eu/debian/.

Flashwalker, 2024-01-06:
> Ah, sorry, missed that. You could try the blueman-git package from https://launchpad.net/~pipewire-debian/+archive/ubuntu/pipewire-upstream for 2.3.5 or my snapshots on https://blueman.cschramm.eu/debian/.

Yeap, thanks!
I just tested and the issue is not present in the blueman-git version

Flashwalker, 2024-01-06:
Oh, no, I jumped to conclusions.

After I turn down the integrated adapter, the blueman-maager window still exists and I can use the USB adapter there.
But only until I minimize the window to tray. After that I can't open window until I bring up the integrated adapter.

closed 2024-03-14 by github-actions[bot]