blueman-project/blueman #2223: Blueman can't find an adapter

aizyuval, 2023-12-16:
blueman: 2.3.5
BlueZ: 5.64-0ubuntu1.1
Distribution: Linux Mint 21.2
Desktop environment: xfce.4.18

[V] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

It usually happens when I boot, or after some period of time, using the pc. And is solved only by reinstalling blueman, each time it happens.(everyday)

This is the output of `blueman-applet --loglevel debug` (than hangs):
```
`blueman-applet 08.17.01 INFO     PluginManager:87 load_plugin: ['DisconnectItems', 'AutoConnect', 'Menu', 'Networking', 'ExitItem', 'ShowConnected', 'GameControllerWakelock', 'DBusService', 'NMPANSupport', 'DiscvManager', 'ConnectionNotifier', 'NMDUNSupport', 'NetUsage', 'KillSwitch', 'StandardItems', 'StatusNotifierItem', 'StatusIcon', 'TransferService', 'SerialManager', 'PowerManager', 'RecentConns', 'AuthAgent', 'DhcpClient', 'PPPSupport']
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 08.17.01 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 08.17.01 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 08.17.01 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 08.17.01 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 08.17.02 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 08.17.02 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.48
blueman-applet 08.17.02 INFO     Applet:63 _on_dbus_name_appeared: org.bluez :1.4
blueman-applet 08.17.02 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 08.17.02 INFO     Networking:71 set_nap   : set nap False
blueman-applet 08.17.02 WARNING  DiscvManager:119 update_menuitems: warning: Adapter is None
blueman-applet 08.17.02 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 08.17.02 INFO     AgentManager:17 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 08.17.02 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
blueman-applet 08.17.03 INFO     PowerManager:140 request_power_state: Requesting False
blueman-applet 08.17.03 INFO     KillSwitch:138 on_power_state_change_requested: False
blueman-applet 08.17.03 DEBUG    KillSwitch:151 on_power_state_change_requested: Using mechanism to set state: False
blueman-applet 08.17.03 INFO     PowerManager:125 check     : callbacks done
blueman-applet 08.17.03 INFO     PowerManager:96 set_adapter_state: False
blueman-applet 08.17.03 INFO     PowerManager:179 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 08.17.03 INFO     PowerManager:182 update_power_state: Signalling False
`
```

When I start **blueman-manager**, it then errors:
```
blueman-manager 08.00.51 ERROR    Manager:147 on_dbus_name_appeared: Default adapter not found, trying first available.
blueman-manager 08.00.51 ERROR    Manager:151 on_dbus_name_appeared: No adapter(s) found, exiting

```
**hci** is also not responsive:
```
hciconfig -a:
Can't get device info: No such device` 
```

`bluetoothctl list` have **no output**.

No **airplane mode** or such:
```
rfkill list
0: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

```

infirit, 2023-12-16:
Are you sure your bluetooth adapter is actually supported by linux? It may also fail to load the driver for some reason, missing firmware perhaps.

aizyuval, 2023-12-18:
> Are you sure your bluetooth adapter is actually supported by linux? It may also fail to load the driver for some reason, missing firmware perhaps.

Yes, it’s supported and works well sometimes.

cschramm, 2023-12-19:
You're saying your hci0 vanishes at some point? Most probably `dmesg` got something for you at that point.

> solved only by reinstalling blueman

That's weird. `bluetoothctl list` does not show you anything, then you `apt-get install --reinstall blueman` and afterward it does?

closed 2024-01-16 by github-actions[bot]