blueman-project/blueman #2226: Blueman cannot detect Samsung Soundbar HW-H450/TK

vgazioglu, 2023-12-18:
blueman: 2.3.5
BlueZ: 5.64
Distribution: Linux Mint 21.2 Victoria base: Ubuntu 22.04 jammy
Desktop environment: Cinnamon 5.8.4 tk: GTK 3.24.33
    wm: muffin dm: LightDM

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Blueman can detect other devices like Samsung mobile phone or bluetooth earbuds but not Soundbar. Issue happened after Linux Mint 21.2 upgrade.

Samsung Soundbar HW-H450/TK
1.Bluetooth Device : [Samsung] Soundbar
2.MAC Address : BC:14:85:D3:XX:XX
3.Type : Classic - BR/EDR
4.Profiles Supported:
A2DP - SNK
AVRCP
5.UUID List :
0000110b-0000-1000-8000-00805f9b34fb
0000110e-0000-1000-8000-00805f9b34fb

infirit, 2023-12-19:
It may be hidden as some devices aren't named initially. In the view menu there is an option to show those. Does it show up then?

vgazioglu, 2023-12-19:
That's not the case I think, please see the attached screenshot
[image]

vgazioglu, 2023-12-19:
Attached is from bluetoothctl

[image]

infirit, 2023-12-19:
If `bc:14:86..` is the address of the soundbar then it's also not available in bluetoothctl. Make the soundbar discoverable/pairable and do `scan on ` in bluetoothctl. Does it show up there now?

vgazioglu, 2023-12-19:
Yes BC:... is Samsung soundbar. It's discoverable (other device home can discover it)
Attached is scan on command syslog
Again 2 other devices are in list and NO SIGN of Samsung Soundbar as if it's not alive - Huh! 

(PS: Also tried agent on/off and 0000)

Is there any way of resetting bluetooth adapter, or may be there's some blocking of UUIDs, or some profile issue, related with Linux Mint sound server system or the RF sgignaling parameters? And may be the reason is the kernel(drivers)  itself 6.2.0-37-generic 

Many thanks!

Dec 19 21:48:18 vegae7250 systemd[1]: systemd-hostnamed.service: Deactivated successfully.
Dec 19 21:48:18 vegae7250 systemd[1]: blueman-mechanism.service: Deactivated successfully.
Dec 19 21:49:27 vegae7250 bluetoothd[117943]: src/agent.c:agent_ref() 0x5561fd454e30: ref=1
Dec 19 21:49:27 vegae7250 bluetoothd[117943]: src/agent.c:register_agent() agent :1.416
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:start_discovery() sender :1.416
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:update_discovery_filter() 
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:discovery_filter_to_mgmt_cp() 
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:trigger_start_discovery() 
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:cancel_passive_scanning() 
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:start_discovery_timeout() 
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 0
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:start_discovery_complete() status 0x00
Dec 19 21:50:09 vegae7250 bluetoothd[117943]: src/adapter.c:discovering_callback() hci0 type 7 discovering 1 method 0
Dec 19 21:50:19 vegae7250 bluetoothd[117943]: src/adapter.c:discovering_callback() hci0 type 7 discovering 0 method 0
Dec 19 21:50:19 vegae7250 bluetoothd[117943]: src/adapter.c:trigger_start_discovery() 
Dec 19 21:50:19 vegae7250 bluetoothd[117943]: src/adapter.c:cancel_passive_scanning() 
Dec 19 21:50:25 vegae7250 bluetoothd[117943]: src/adapter.c:start_discovery_timeout() 
Dec 19 21:50:25 vegae7250 bluetoothd[117943]: src/adapter.c:start_discovery_timeout() adapter->current_discovery_filter == 0
Dec 19 21:50:25 vegae7250 bluetoothd[117943]: src/adapter.c:start_discovery_complete() status 0x00
Dec 19 21:50:25 vegae7250 bluetoothd[117943]: src/adapter.c:discovering_callback() hci0 type 7 discovering 1 method 0

infirit, 2023-12-19:
BlueZ caches things in `/var/lib/bluetooth` with a directory per device address (BC:...). If there is is one for the soundbar you can try deleting (as root) it and try gain.

vgazioglu, 2023-12-20:
Yea I found it under /var/lib/bluetooth/18:5E:0F:A8:XX:XX/cache and deleted it as root. Nothing changed :(

vgazioglu, 2023-12-20:
Today I decided to move the soundbar very close to the laptop (actually from original appr 5 meters away) and connection established.
My conlusion is there there's an interference in the room, possible with some 2.4 GHz Wifi.
My problem has not benn resolved but I confirm that you can close the issue.
Sorry for taking your valuable time, thanks and best regards infirit

infirit, 2023-12-21:
No worries. For me 1 thick concrete wall can cause choppy bluetooth with my usb dongles already. Especially noticible when doing audio.

closed 2023-12-21 by infirit