blueman-project/blueman #2227: Bluetooth connection with Brydge 10.5 keyboard

baqlias, 2023-12-20:
blueman: 2.3.5
BlueZ:
Distribution: Linux Mint 21.2
Desktop environment: Microsoft Surface Go 2, M3 CPU

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

baqlias, 2023-12-20:
For some reason, my comment didn't posy above, so I'll add it here.  I have been trying to link a Bridge 10.5 Bluetooth keyboard to my Surface Go 2 which now runs Linux Mint 21.2.  Blueman correctly identifies the keyboard. When I try to connect, I get a message asking me to enter a 6-digit code. I do so, but there's no indication that my keyboard's strokes are registering at all. I get a message that I am connected, but the connection then drops in a few seconds.  

Some say that Brydge keyboards aren't compatible with Linux. If you know that this is the case, that answers my question. If not, let me know what logs I can post.  Thanks!

infirit, 2023-12-20:
> For some reason, my comment didn't posy above

You posted between ``, I fixed it for you. Can you post a blueman-applet log as described here https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman.

You may need to press enter to complete the pin.

baqlias, 2023-12-20:
[bluemanlog.odt](https://github.com/blueman-project/blueman/files/13732103/bluemanlog.odt)
The keyboard works now, but the touchpad doesn't work at all.

baqlias, 2023-12-20:
[xinput.odt](https://github.com/blueman-project/blueman/files/13732156/xinput.odt)
The touchpad is recognized as a mouse, apparently!

cschramm, 2024-01-01:
Initial problem gone. Broken touchpad could be an issue in BlueZ 5.64 (https://github.com/bluez/bluez/issues/298). I guess you're stuck with that version on Linux Mint 21. You could try https://packages.ubuntu.com/lunar/bluez.

closed 2024-03-10 by github-actions[bot]