blueman-project/blueman #2229: bluetooth is not working

ShamanK18, 2023-12-26:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

something is not working in my bluetooth as you can see that even after the search option no device 
[image]
[image]

cschramm, 2024-01-01:
Assuming that you're able to discover some devices from another device at the same time, check if it might be hidden due to no name provided in the "View" menu.

ShamanK18, 2024-01-04:
> Assuming that you're able to discover some devices from another device at the same time, check if it might be hidden due to no name provided in the "View" menu.

I checked everything sir but no device is showing nor is the name of the PC showing in the phone.

infirit, 2024-01-04:
Iirc, the Bluetooth adapter is hidden by default, open blueman-adapters to change this.

ShamanK18, 2024-01-05:
> Iirc, the Bluetooth adapter is hidden by default, open blueman-adapters to change this.

I didn't see anything else other than the screenshots I have given above.

ShamanK18, 2024-01-05:
> Iirc, the Bluetooth adapter is hidden by default, open blueman-adapters to change this.

I have checked all the settings

cschramm, 2024-01-05:
Not sure I can follow. You have an unknown version of blueman running on an unknown system and try to discover another PC? You do see that PC with another device that it's not paired with - say, your smartphone -, so it is discoverable, but you do not see it in blueman on the unknown system? Are you able to discover other devices - say, your smartphone? Do you see discoverable devices in other tools on the unknown system - say, bluetoothctl?

ShamanK18, 2024-01-06:
> Not sure I can follow. You have an unknown version of blueman running on an unknown system and try to discover another PC? You do see that PC with another device that it's not paired with - say, [...]

Check this, whatever you want to check, check it and then suggest a solution to fix it.[
[image]
](url
![Uploading 20240106_170755.jpg…](
[image]
[image]
[image]
[image]
[image]
[image]

cschramm, 2024-01-06:
I cannot conclude anything I asked for from those photos, just that it seems to be some blueman 2.3, that you are hiding unnamed devices and that the adapter is set discoverable.

I'm still unsure if you're trying to discover some device in blueman or to make the system discoverable with blueman. The latter seems to work fine. If you cannot actually discover it, double check the procedure and possibly try from a different device. In case of issues with your adapter, I'd expect some kernel logs.

ShamanK18, 2024-01-09:
> I cannot conclude anything I asked for from those photos, just that it seems to be some blueman 2.3, that you are hiding unnamed devices and that the adapter is set discoverable.
>
> I'm still unsure [...]

I have to connect bluetooth ear buds to it please help how can I fix it

> I cannot conclude anything I asked for from those photos, just that it seems to be some blueman 2.3, that you are hiding unnamed devices and that the adapter is set discoverable.
>
> I'm still unsure [...]

cschramm, 2024-01-09:
So you want to discover ear buds from a system that runs blueman 2.3 (I'm ignoring that you talked about a PC not being visible on a phone :shrug:).

First check that the ear buds are actually discoverable. You can use your phone to search for them (make sure they are not already known to the phone initially). If you cannot find them there either, the conclusion is that they are not discoverable. Consult their manual or support on how to get them paired.

In case they do show up on the phone but not in blueman when searching at the same time and distance, run `bluetoothctl` in a terminal window on the system that you're running blueman on, enter `scan on` in it and see if they do show up there. If not, you successfully ruled out an issue with blueman. In case no devices show up at all, you might have a problem with the system's adapter or driver and there might be kernel logs regarding it.

ShamanK18, 2024-01-10:
> So you want to discover ear buds from a system that runs blueman 2.3 (I'm ignoring that you talked about a PC not being visible on a phone 🤷).
> 
> First check that the ear buds are actually discoverable. You can use your phone to search for them (make sure they are not already known to the phone initially). If you cannot find them there either, the conclusion is that they are not discoverable. Consult their manual or support on how to get them paired.
> 
> In case they do show up on the phone but not in blueman when searching at the same time and distance, run `bluetoothctl` in a terminal window on the system that you're running blueman on, enter `scan on` in it and see if they do show up there. If not, you successfully ruled out an issue with blueman. In case no devices show up at all, you might have a problem with the system's adapter or driver and there might be kernel logs regarding it.

I have tried all this but it is not showing anything, neither is the name of bluetooth of my phone nor of ear buds shown but it works well in windows but there also after updating the bluetooth driver, bluetooth is not working. It is working but nothing is working here nor did I find the option to update the bluetooth driver anywhere.

cschramm, 2024-01-10:
So you can discover the ear buds from a Windows system but at the same time neither from the system running blueman nor from some phone? The ear buds might be coupled to the Windows system then, although the Windows system does not have a pairing anymore. Consult their manual or support on how to get them decoupled.