blueman-project/blueman #2233: Blueman not showing up in Gnome launcher as a GUI application and doesn't follow system's dark mode theme

Grausiga, 2024-01-02:
blueman: 2.3.5
BlueZ: not sure
Distribution: Debian 12 (stable)
Desktop environment: Gnome 43.9

* [ - ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The application works as intended, par the mentioned problem with it not showing up in the Gnome launcher and not being in dark mode - I have to start it through the terminal. It was installed from the stable debian repo.

Thought I'd mention it here, as it worked flawlessly with the same machine in Ubuntu 22.04-23.10 until I switched to pure Debian.

infirit, 2024-01-02:
To be clear, switching light/dark theme on Ubuntu changes as expected but it doesn't on Debian 12?

Regardless, we don't do anything special and rely on Gtk+ to handle theme changes.

SamPurple22, 2024-01-11:
Seeing the same issue on Debian 12 with GNOME 43.9, where after installing the blueman package, it's desktop entries do not show up in Gnome Shell.

However, just renaming /usr/share/applications/blueman-manager.desktop to /usr/share/applications/b_m.desktop for example, results in a visible "Bluetooth Manager" entry in the Applications grid or when searching for apps.

cschramm, 2024-01-11:
Yeah, well, Debian ships an `/etc/gnome/menus.blacklist` in `gnome-menus` that includes `blueman-manager.desktop`: https://salsa.debian.org/gnome-team/gnome-menus/-/blob/debian/latest/debian/menus.blacklist?ref_type=heads :shrug: It seems like that's been the case already since Debian 9.

Grausiga, 2024-01-14:
> Seeing the same issue on Debian 12 with GNOME 43.9, where after installing the blueman package, it's desktop entries do not show up in Gnome Shell.
> 
> However, just renaming /usr/share/applications/blueman-manager.desktop to /usr/share/applications/b_m.desktop for example, results in a visible "Bluetooth Manager" entry in the Applications grid or when searching for apps.
> 
> Dark theme works after setting this to 1:
> 
> ```
> cat ~/.config/gtk-3.0/settings.ini
> 
> [Settings]
> gtk-application-prefer-dark-theme=1
> ```

Thank you!

closed 2024-01-14 by Grausiga