blueman-project/blueman #2234: Install Issues Not Finding Python libs

tthompsongmsi, 2024-01-02:
blueman: 2.3.5
BlueZ: 5.64-2.el9.x86_64
Distribution: Rocky Linux 9
Desktop environment: Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Did not find topic on python issues. I did try to set the PKG_CONFIG_PATH environment variable to where my python site-packages exist but that did not work

PKG_CONFIG_PATH = "~/.local/lib/python3.9/site-packages"

checking for python version... 3.9
checking for python platform... linux
checking for GNU default python prefix... ${prefix}
checking for GNU default python exec_prefix... ${exec_prefix}
checking for python script directory (pythondir)... ${PYTHON_PREFIX}/lib/python3.9/site-packages
checking for python extension module directory (pyexecdir)... ${PYTHON_EXEC_PREFIX}/lib64/python3.9/site-packages
checking for python-3.9... no
configure: error: Package requirements (python-3.9) were not met:

Package 'python-3.9', required by 'virtual:world', not found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables PYTHON_CFLAGS
and PYTHON_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
❯ python --version
Python 3.9.18
❯ which python
/usr/bin/python

cschramm, 2024-01-02:
The Python installation typically does not contain development files, including pkg-config data. You're probably lacking some `-devel` package like `python3-devel` or `python39-devel` for that discovery to work.

You're entering a rocky road (no pun intended) and I guess you're better off using [EPEL](https://docs.fedoraproject.org/en-US/epel/) to enjoy Fedora's blueman package.

infirit, 2024-01-02:
I think Leigh mentioned it was never in EPEL, was it added recently?

My recommendation if it's not in EPEL is to download the src.rpm for Fedora and build it for rocky 9 with rpmbuild/mock. Doing a manual build is not as straightforward as it may seem.

cschramm, 2024-01-02:
Ah, true, #786, forgive my Enterprise world confusion. :see_no_evil:

tthompsongmsi, 2024-01-02:
> Curious, how did you even get this far as we force automake 1.16.3 and
> that's not available for RHEL9 😕

I manually installed that version to get past that. I'm also having issues
where bluez is installed but it can not be found when running ./configure

Thanks,

Trevor Thompson
Member of Technical Staff
Green Mountain Semiconductor
(802)503-7125

infirit, 2024-01-02:
I have a patch and modified spec file for you. You will need to:

* setup EPEL
* setup rpmbuild dev tree 
* save the modified spec file to `rpmbuild/SPEC`
* extract the sources from  `blueman-2.3.5-8.fc40.src.rpm` to `rpmbuild/SOURCES` with something like file-roller (blueman-2.3.5.tar.gz and 4 patched)
* copy the fifth patch for automake 1.16.2 (1.16.3 is required for python3.10 which is not relevant for RHEL as it's on 3.9)
* run `rpmbuild -bb /path/to/rpmbuild/dir/SPEC/blueman.spec`

This will spit out a set of rpm file in `rpmbuild/RPMS` that you can install. The base package is in the `x86-64`

ZIP with patch and spec.
[blueman-rhel-spec-patch.zip](https://github.com/blueman-project/blueman/files/13813631/blueman-rhel-spec-patch.zip)

You may want to ask the EPEL folks nicely to add blueman, it requires only the downgrade to automake-1.16.2.

tthompsongmsi, 2024-01-02:
Thank you, I will give this a try and report back.

Thanks,

Trevor Thompson
Member of Technical Staff
Green Mountain Semiconductor
(802)503-7125

closed 2024-03-11 by github-actions[bot]