blueman-project/blueman #2236: Change device icon

burtek, 2024-01-05:
blueman: 2.3.5
BlueZ:
Distribution: LMDE 6
Desktop environment: Cinnamon 5.8.4

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Is there a way to change paired device icon? I have a lot of different audio devices paired, all of them share same headset icon even though they might be a speaker with mic and would like to replace at least some icons with some other icon/picture to be able to more easily tell them apart

cschramm, 2024-01-05:
There currently is not. Icon names are provided by bluetoothd, probably `audio-headphones` in that case and we just use those icon names in the GUI. A backend to override them would not be a problem - you could use e.g. `dconf` to control it -, but a user-friendly interface would probably require an icon picker (instead of just an input for an icon name).

burtek, 2024-01-05:
How can I change the icon using `dconf`? Can't find any instructions on that and I'm not `dconf`-familiar :/

infirit, 2024-01-07:
You can't. @cschramm was thinking ahead on how this _could_ be implemented.

burtek, 2024-01-07:
Oh, gotcha. Thanks for clarification

AndreiMiculita, 2024-01-22:
While on the topic, would it make sense to change the icon for Loudspeaker devices to something more appropriate? The current icon showing a literal circuit board is frankly a bit ugly and not very user-friendly.

[image]

Using the Adwaita icon theme but it looks very similar in most themes.