blueman-project/blueman #2237: Disable 'NMDUNSupport' and 'NMPANSupport' and not showing the applet when clicking in waybar tray icon.

mrpingo, 2024-01-05:
blueman: 2.3.5
BlueZ: 5.70
Distribution: Gentoo
Desktop environment:  Wayland

Hello and many thanks for all the effort.

Is it possible that I had missing something on the way, but I have this error when clicking in the tray blueman-applet in Waybar and the window is also not shown. The error I get is shown in the next lines:

`Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/DBusProxies.py", line 52, in call_finish
    proxy.call_finish(resp)
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying (4)`

Also getting this odd NetworkManager error when launching it as well because I don't have it and I don't need it and don't want to install it:

`blueman-applet 19.36.30 ERROR    PluginManager:92 load_plugin: Unable to load plugin module NMDUNSupport
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/NetworkManager.py", line 9, in <module>
    gi.require_version('NM', '1.0')
  File "/usr/lib/python3.11/site-packages/gi/__init__.py", line 126, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace NM not available`

Is it possible to disable this 'NMDUNSupport' and 'NMPANSupport' in config or in any way?

By the way, I am also getting this warning every now and then (each 4s to 30s) in waybar, redirected log to sway.log:

`(waybar:7628): LIBDBUSMENU-GLIB-WARNING **: 19:58:15.399: Unable to replace properties on 0: Error getting properties for ID
`
Is it a waybar bug?

Sorry for all this mess in a single issue, should I split every issue?

infirit, 2024-01-07:
> Is it possible that I had missing something on the way, but I have this error when clicking in the tray blueman-applet in Waybar and the window is also not shown. The error I get is shown in the [...]

Please post the full debug log of `blueman-applet --loglevel debug` and describe the exact steps how you got to this Traceback.

> Also getting this odd NetworkManager error when launching it as well because I don't have it and I don't need it and don't want to install it:
>
> `blueman-applet 19.36.30 ERROR PluginManager:92 [...]

This is expected if you haven't installed NetworkManager gobject imtrospection package. It's harmless.

> Is it possible to disable this 'NMDUNSupport' and 'NMPANSupport' in config or in any way?

No and there is no reason to disable it.

> By the way, I am also getting this warning every now and then (each 4s to 30s) in waybar, redirected log to sway.log:
> 
> `(waybar:7628): LIBDBUSMENU-GLIB-WARNING **: 19:58:15.399: Unable to replace properties on 0: Error getting properties for ID ` Is it a waybar bug?

Maybe? Better ask them about it.

cschramm, 2024-01-10:
I put a :moneybag: on https://github.com/blueman-project/blueman/issues/2158#issuecomment-1751947232 and am curious to read about the specific cause in this case.

mrpingo, 2024-01-10:
You are the absolutely winner of the 💰 ; ) , this D-Bus issue came because it started as a D-Bus service in OpenRC, I really don't know how dbus works and I have been fighting with it for ages. It seems you have to not start it in anyway and -it manages itself when required?-  I really want to know how to manage DBus..
Just for the the record in case anyone else is facing this same issue. Removing the service when the computer starts seems to solve the problem, and starts the applet  within the other errors in the logs, but not those related with D-Bus and the manager opens and closes at every click. So I will put this thread as closed since every issue in it was adressed.

Many thanks to all. Have a beautiful week.

Regards

closed 2024-01-10 by mrpingo