blueman-project/blueman #2242: compiled but not found, ModuleNotFoundError: No module named 'blueman'

abdelouahabb, 2024-01-28:
blueman: Latest branche / 2.3.5
BlueZ:
Distribution:
Desktop environment: Gnome

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hello,
I try to compile it locally on `Pop!_OS 22.04 LTS using Gnome 42.9`, everything went well, make and make install worked without errors, but I still can't find it hitting me with the error : 

Traceback (most recent call last):
  File "/usr/local/bin/blueman-applet", line 16, in <module>
    from blueman.Functions import create_logger, create_parser, set_proc_title
ModuleNotFoundError: No module named 'blueman'

[image]

infirit, 2024-01-28:
Just use blueman that is already packaged by ubuntu.

cschramm, 2024-01-28:
You installed to `/usr/local`. Assuming you're using a system provided Python installation, it is not configured to look for packages there.

abdelouahabb, 2024-01-28:
> Just use blueman that is already packaged by ubuntu.

The one with 22.04 does not offer battery status

abdelouahabb, 2024-01-28:
> You installed to `/usr/local`. Assuming you're using a system provided Python installation, it is not configured to look for packages there.

I guess it's a better idea to retrieve Python env and install it there ?

abdelouahabb, 2024-01-29:
Here is what happens about Python,
This is the default behavior : 
`./configure PYTHON=/usr/bin/python`

To understand which python it seems, I did : 
`./configure PYTHON=/usr/local/bin/python3`

→ ```
blueman-manager 
Traceback (most recent call last):
  File "/usr/local/bin/blueman-manager", line 15, in <module>
    from blueman.main.Manager import Blueman
  File "/usr/local/lib/python3.11/site-packages/blueman/main/Manager.py", line 6, in <module>
    from blueman.bluez.Adapter import Adapter
  File "/usr/local/lib/python3.11/site-packages/blueman/bluez/Adapter.py", line 3, in <module>
    from gi.repository import GLib
ModuleNotFoundError: No module named 'gi'
```

So why blueman is not installed into the default python ?

cschramm, 2024-01-29:
You need to `./configure` at least `--prefix /usr` if you want to spread it over your system. ...or use a custom Python that also loads from `/usr/local` if you install it there. `PYTHON=/usr/local/bin/python3` would be correct if you actually have one there.

Many other things will break, e.g. you need to configure correct systemd paths, see https://salsa.debian.org/debian/blueman/-/blob/dd9b5402dbc101b836c19966c9be2a1b43992588/debian/rules#L9. The better idea would be to build a Debian package. See that salsa repository for configuration. Or just try a snapshot from https://blueman.cschramm.eu/debian/ where that's already done.

abdelouahabb, 2024-01-29:
Now it seems i get an error with dbus, even if I installed `dbus-x11`
[image]

cschramm, 2024-01-29:
If you did not uninstall what you spread over your system previously, you might have e.g. competing `org.blueman.Manager.service` in `/usr/share/dbus-1/services` (package managed) and `/usr/local/share/dbus-1/services` (configured and installed by you).