blueman-project/blueman #2246: Signal strength tooltip is unclear

sboukortt, 2024-02-01:
blueman: 2.3.5

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The tooltip that indicates signal strength is saying that it’s “Too much”:

[image]

I can’t tell for sure whether it’s trying to be tongue-in-cheek or whether there really is such a thing as “too much”. As far as I can tell, the tooltip was introduced 15 years ago with no explanation: https://github.com/blueman-project/blueman/commit/32ac928f23a043013e6d14aa215cd8446af40d19

Would it be possible to make this clearer?

Thanks.

infirit, 2024-02-01:
Some historical background why it is like this. For many (if not all) bluetooth 5/le devices signal information is not available like it was before. And because we otherwise have little to no indication otherwise to show a device is connected I introduced this ugly hack some time ago. It's made intentionally obvious this is fake data and not meant to be tongue-in-cheek but I can see how you might think it was 😄 .

cschramm, 2024-02-01:
@infirit: Those terms predate our work. :smile:

I agree that they make very limited sense. Actually you cannot really interpret the RSSI of an arbitrary device.

This falls into the cluster of signal indicator issues that I recently moved to milestone 2.5 to no longer delay a 2.4 feature release.