blueman-project/blueman #2248: Keyboard connects and disconnects rapidly

dalb8, 2024-02-02:
blueman: 2.3.5+mint1+vera
BlueZ: 5.64-0ubuntu1.1
Distribution: Mint 21.3
Desktop environment: Cinnamon

Hi! I just fresh-installed the new version of Linux Mint and the cheap folding keyboard I have connects and disconnect rapidly. It was working with the previous Mint. In red it says br-connection-failed socket.

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Here is the log from the applet:
https://gist.github.com/dalb8/1aa2be4a40b89ea967997b483c33d6c0

cschramm, 2024-02-04:
Did you try to explicitly pair the device instead of directly connecting to it?

dalb8, 2024-02-05:
Thanks for the tip. It didn't work at first but when I took the keyboard out of discoverable mode a notification appeared asking should I accept the pairing and when I clicked that the keyboard works.