blueman-project/blueman #2251: Logitech K810 - does not pair

W2ARZ, 2024-02-03:
blueman: 2.3.5
BlueZ:  
Distribution:  LinuxMint 21.3
Desktop environment:  Cinnamon

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Unable to pair Logitech K810 keyboard with fresh LinuxMint 21.3 install.   Blueman does not even seem to find the K810.  Reviewed open issue with Logitech K380 keyboard, and not the same.

cschramm, 2024-02-04:
It might be hidden in case it does not provide a name. Check the manager window's "View" menu.

W2ARZ, 2024-02-06:
Had the "Hide unnamed devices" unchecked and it was not there.  No changes made, and today, it's finding my keyboard, but I'm getting error:

Connection Failed: br-connection-create-socket

So, progress is being made, but will keep working on it.

W2ARZ, 2024-02-09:
Closing.  Popup appeared later the day of previous post with 6 digit pairing code.  Been working perfectly since.  Wish I could say what changed.  Thank you all.

closed 2024-02-09 by W2ARZ