blueman-project/blueman #2254: GUI Gamepad connection problems

fasddsaf, 2024-02-05:
blueman: 2.3.5
BlueZ: 5.64-0ubuntu1.1
Distribution: Linux Mint 21.3
Desktop environment: Cinnamon 6.0.4

Sorry, I don't speak English and used Google translator.

If I connect using blueman-manager, the gamepad buttons do not work. After turning off the gamepad, it does not connect again and disappears from the list of paired devices.

If I use bluetoothctl, the gamepads connect and work.

The audio device connects and works.

infirit, 2024-02-05:
If you choose "Connect" in blueman-manager try using the "Pair" option first. Does it work properly then?

If that doesn't help please send the blueman-applet debug log starting with the pair attempt. See https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

fasddsaf, 2024-02-05:
Yes, if I use the "Pair" option first and then the "Connect" option,
which becomes available after about 2 seconds, then everything works
properly.

One time when I was connecting using bluetoothctl I entered "pair" and
then "connect" very quickly and the connection did not work correctly.
The buttons did not work and there was no automatic reconnection.
However, this may not mean anything, because... I was unable to repeat
this and the next attempts everything worked correctly.

$ blueman-applet --loglevel debug
blueman-applet 19.15.36 INFO     PluginManager:87 load_plugin:
['PowerManager', 'DisconnectItems', 'TransferService', 'NMDUNSupport',
'AuthAgent', 'StatusIcon', 'Menu', 'PPPSupport', 'StandardItems',
'AutoConnect', 'DBusService', 'NetUsage', 'ShowConnected',
'DiscvManager', 'DhcpClient', 'StatusNotifierItem',
'ConnectionNotifier', 'KillSwitch', 'GameControllerWakelock',
'RecentConns', 'Networking', 'NMPANSupport', 'SerialManager',
'ExitItem']
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 19.15.36 DEBUG    Base:60 do_g_properties_changed:
/org/bluez/hci0 {'Address': '18:47:3D:E1:BD:1A', 'AddressType':
'public', 'Name': 'Computer', 'Alias': 'Computer', 'Class': 8126732,
'Powered': True, 'Discoverable': False, 'DiscoverableTimeout': 0,
'Pairable': True, 'PairableTimeout': 0, 'Discovering': True, 'UUIDs':
['00001133-0000-1000-8000-00805f9b34fb',
'0000110e-0000-1000-8000-00805f9b34fb',
'00001105-0000-1000-8000-00805f9b34fb',
'00001132-0000-1000-8000-00805f9b34fb',
'00001200-0000-1000-8000-00805f9b34fb',
'00001104-0000-1000-8000-00805f9b34fb',
'00005005-0000-1000-8000-0002ee000001',
'00001108-0000-1000-8000-00805f9b34fb',
'0000110c-0000-1000-8000-00805f9b34fb',
'00001801-0000-1000-8000-00805f9b34fb',
'0000112f-0000-1000-8000-00805f9b34fb',
'0000180a-0000-1000-8000-00805f9b34fb',
'0000110b-0000-1000-8000-00805f9b34fb',
'00001800-0000-1000-8000-00805f9b34fb',
'0000111f-0000-1000-8000-00805f9b34fb',
'0000110a-0000-1000-8000-00805f9b34fb',
'00001106-0000-1000-8000-00805f9b34fb'], 'Modalias':
'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 19.15.36 WARNING  PluginManager:151 __load_plugin: Not
loading PPPSupport because its conflict has higher priority
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 19.15.36 INFO     PluginManager:160 __load_plugin:
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 19.15.36 WARNING  PluginManager:151 __load_plugin: Not
loading DhcpClient because its conflict has higher priority
[... 155 lines cut ...]
<blueman.plugins.applet.NMDUNSupport.NMDUNSupport object at
0x7fcc3caf1000>
blueman-applet 19.15.36 DEBUG    BasePlugin:63 _on_plugin_delete:
Deleting plugin instance
<blueman.plugins.applet.DBusService.DBusService object at
0x7fcc3caf11e0>
blueman-applet 19.15.36 DEBUG    BasePlugin:63 _on_plugin_delete:
Deleting plugin instance
<blueman.plugins.applet.TransferService.TransferService object at
0x7fcc3caf1240>
blueman-applet 19.15.36 DEBUG    BasePlugin:63 _on_plugin_delete:
Deleting plugin instance
<blueman.plugins.applet.DisconnectItems.DisconnectItems object at
0x7fcc3caf0f70>
blueman-applet 19.15.36 DEBUG    BasePlugin:63 _on_plugin_delete:
Deleting plugin instance
<blueman.plugins.applet.PowerManager.PowerManager object at
0x7fcc3caf0f10>
blueman-applet 19.15.36 DEBUG    BasePlugin:63 _on_plugin_delete:
Deleting plugin instance <StatusIcon.StatusIcon object at
0x7fcc410eae00 (blueman+plugins+applet+StatusIcon+StatusIcon at
0x560cfb960cc0)>
blueman-applet 19.15.36 DEBUG    BasePlugin:63 _on_plugin_delete:
Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at
0x7fcc3cb5ceb0>

ihatetoregister, 2024-06-02:
Had the same issue, and doing a pair created /dev/input/js0. Just doing a connect did not. Maybe this could be improved?