blueman-project/blueman #2255: ps3 controller trust reset on USB reconnection

twistedddx, 2024-02-05:
blueman: 2.3.5
BlueZ: 5.64
Distribution: Linux Mint
Desktop environment: xcfe

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Everytime I connect USB to a genuine PlayStation 3 controller a "Bluetooth Authentication" pops asking for Always Accept, accept, debt.
I click always accept and the device is shown in blueman Bluetooth Devices with a green tick.
If I disconnect USB and reconnect I am again greeted with Bluetooth Authentication window asking if to always accept, accept or deny. This occurs everytime the USB is reconnected.

I believe this issue was an issue in bluez and was corrected here:
https://github.com/bluez/bluez/issues/372
https://patchwork.kernel.org/project/bluetooth/patch/20220802163441.2522333-1-luiz.dentz@gmail.com/

After a user has manually selected to always accept/trust the device mac that setting should be respected and no future popups are required.

I also have already set:
/etc/bluetooth/input.conf ClassicBondedOnly=false
Which is related to being able to bond to the ps3 controller.

bluetooth.service logs this on USB connection:
bluetoothd: sixaxis: compatible device connected. Sony PLAYSTATION(R)3 Controller
bluetoothd: sixaxis: setting up new device

twistedddx, 2024-02-05:
Blueman-applet debug log:
blueman-applet 21.33.31 DEBUG    Manager:52 _on_object_added: Device1: /org/bluez/hci0/dev_<snip>
blueman-applet 21.33.31 INFO     Applet:95 on_device_created: Device created /org/bluez/hci0/dev_<snip>
blueman-applet 21.33.31 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0/dev_<snip> {'Address': '<snip>', 'AddressType': 'public', 'Name': 'Sony PLAYSTATION(R)3 Controller', 'Alias': 'Sony PLAYSTATION(R)3 Controller', 'Paired': False, 'Trusted': False, 'Blocked': False, 'LegacyPairing': False, 'Connected': False, 'UUIDs': [], 'Modalias': 'usb:v054Cp0268d0000', 'Adapter': '/org/bluez/hci0', 'ServicesResolved': False}
blueman-applet 21.33.31 INFO     BluezAgent:237 _on_authorize_service: Agent.Authorize

infirit, 2024-02-05:
Blueman is asked to handle authentication and we do. The request is send by BlueZ and we have no control over it.

input.conf should look like this BTW.
```
[General]
ClassicBondedOnly=false
```

twistedddx, 2024-02-05:
I confirm my input.conf does already is as you said.
If ClassicBondedOnly=true, which is the current default, a screen asking for pairing pin opens and the Bluetooth can not connect at all.

There is bluez source located in:
/blueman/bluez
It appears this lacks some patches that the bluez repo has. Including the patch that apparently fixed the trust flipping.
Is that source code used?

cschramm, 2024-02-05:
As you mentioned correctly, your distribution provides BlueZ 5.64. That does not contain the change that you mention.

twistedddx, 2024-02-06:
I see the Linux Mint BlueZ is just out of date.
I might continue down the path of compiling bluez from source but a temp hack that appears to work is to grab 5.66 deb from Debian repo.

twistedddx, 2024-02-09:
I am happy for this issue to be closed as it is as simple as BlueZ 5.64 included in the current Linux Mint 21.3 and Ubuntu 22.04 LTS does not include the fix for this exact issue. My bad for assuming that patch would be included by now and that blueman had an issue.

If anyone ends up finding this issue to try and fix it thsemselves this is the steps I did to build and install BlueZ from source on Linux Mint.

Remove bluez 5.64 and install bluez build dependencies:
```
sudo apt remove bluez
sudo apt install git python3-dev python-gi-dev cython3 libbluetooth-dev
```

Point cython at cython3:
```
cd /bin/
sudo nano cython
```
add and save this to the file: cython3 $@
Then make the file executable:
`sudo chmod 755 cython`

Download and build bluez and move bluetoothd to sbin:
```
cd ~
git clone https://github.com/bluez/bluez.git
cd bluez
./configure --prefix=/usr --mandir=/usr/share/man --sysconfdir=/etc --localstatedir=/var --enable-sixaxis
make
sudo make install
sudo cp ./src/bluetoothd /usr/sbin/bluetoothd
```

Use equivs to tell package manager that you have an equivalent to bluez being installed so that you can reinstall blueman and pulseaudio-module-bluetooth:
```
sudo apt install equivs
equivs-control bluez
nano bluez
```
Fill out fields Section, Priority, Standards-Version, Package, Maintainer, Version, Architecture, Description
```
equivs-build bluez
sudo dpkg -i bluez_5.72_all.deb
```
Reinstall blueman and pulseaudio-module-bluetooth
`sudo apt install blueman pulseaudio-module-bluetooth`

Enable Bluetooth bonding via USB:
`nano /etc/bluetooth/input.conf`
Under [General] set ClassicBondedOnly=false

Restart bluetooth
```
sudo systemctl daemon-reload
sudo systemctl restart bluetooth
```

twistedddx, 2024-02-09:
Apparently I can close

closed 2024-02-09 by twistedddx