blueman-project/blueman #2256: Before new version my PS4 controller dont work

alexrowlingofc, 2024-02-07:
blueman: last version
BlueZ:
Distribution:
Desktop environment: dell optlex linux mint 21.3

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

alexrowlingofc, 2024-02-07:
Before updating the Bluetooth manager, I used my PS4 controller on my notebook normally but after updating the controller connects but no buttons work

cschramm, 2024-02-07:
As far as I'm aware, there exist no updates for blueman in Linux Mint. You probably updated the bluez package instead.

infirit, 2024-02-07:
This happens to me when the kernel got updated and I haven't rebooted yet.

emonteirotcc, 2024-03-05:
The same here, my PS4 Controller (generic) is connected with the bluetooth, but not working, none joystick manager are recognizing. But the joystick is working normally connected by cable usb.

gustavolinux, 2024-04-21:
> The same here, my PS4 Controller (generic) is connected with the bluetooth, but not working, none joystick manager are recognizing. But the joystick is working normally connected by cable usb.

I'm also having the same problem here on archlinux(my system is all updated). I'm not able to use any of my bluetooth gamepads and I've tested a bunch of them(all used to work fine weeks ago)

closed 2024-06-28 by github-actions[bot]