blueman-project/blueman #2259: [Question] Does controller turn off turn off the devices?

kuraga, 2024-02-12:
Good day!

On Android, my bluetooth headphones are turned off on "Turn off the bluetooth controller" action.

But on BlueZ & and Blueman -Linux, they aren't turned off. The headphones just lose the signal.

Is this BlueZ or Blueman decision? If the latter is, is it correct?

Thanks!

infirit, 2024-02-12:
When you **disable** bluetooth from the applet icon blueman will set the bluetooth adapter's power state to Off.

Does the headphone turn off if you manually disconnect in blueman manager?

kuraga, 2024-02-12:
> Does the headphone turn off if you manually disconnect in blueman manager?

Yes.

> set the bluetooth adapter's power state to Off.

Should each of the headphones be disconnected _before this action_, shouldn't them?

infirit, 2024-02-12:
Just checked bluedevil and it also just powers down the adapter without disconnecting everything first. We could disconnect devices before powering down if the user deliberately chose to do this. I'll mark this issues as enhancement.