blueman-project/blueman #2270: Blueman search not showing my devices, also my Linux Mint machine not appearing on my phone

RubenFalcao14, 2024-02-20:
My Blueman search is not showing my devices, also my Linux Mint laptop is not appearing on my phone. When I was using Windows 10, bluetooth was working. Please help.

```
dell@DELL:~$ blueman-applet --loglevel debug 
blueman-applet 20.48.04 INFO     PluginManager:87 load_plugin: ['StatusIcon', 'ShowConnected', 'PPPSupport', 'SerialManager', 'ConnectionNotifier', 'NetUsage', 'ExitItem', 'RecentConns', 'AuthAgent', 'NMDUNSupport', 'TransferService', 'StandardItems', 'DBusService', 'StatusNotifierItem', 'Networking', 'AutoConnect', 'KillSwitch', 'PowerManager', 'NMPANSupport', 'Menu', 'DhcpClient', 'DiscvManager', 'DisconnectItems', 'GameControllerWakelock']
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 20.48.04 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': 'AC:D1:B8:CA:ED:CA', 'AddressType': 'public', 'Name': 'DELL', 'Alias': 'DELL', 'Class': 0, 'Powered': False, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['00001133-0000-1000-8000-00805f9b34fb', '0000110e-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00005005-0000-1000-8000-0002ee000001', '00001108-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0540', 'Roles': ['central', 'peripheral']}
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 20.48.04 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem'>
blueman-applet 20.48.04 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 20.48.04 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 20.48.05 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 20.48.05 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 20.48.05 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 20.48.05 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 20.48.05 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 20.48.05 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 20.48.05 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 20.48.05 INFO     KillSwitch:106 io_event  : killswitch registered 3
blueman-applet 20.48.05 INFO     KillSwitch:122 io_event  : State: False
blueman-applet 20.48.05 INFO     PowerManager:179 update_power_state: off True | foff False | on True | current state True | new state False
blueman-applet 20.48.05 INFO     PowerManager:182 update_power_state: Signalling False
blueman-applet 20.48.05 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 20.48.05 INFO     TransferService:234 _on_dbus_name_appeared: org.bluez.obex :1.63
blueman-applet 20.48.05 INFO     Applet:63 _on_dbus_name_appeared: org.bluez :1.4
blueman-applet 20.48.05 INFO     Functions:114 launch    : Gtk eventtime is 0, not using LaunchContext
blueman-applet 20.48.05 INFO     BluezAgent:56 register_agent: Register Agent
blueman-applet 20.48.05 INFO     Networking:71 set_nap   : set nap False
blueman-applet 20.48.05 ERROR    AgentManager:20 on_register_failed: /org/bluez/obex/agent/blueman org.bluez.obex.Error.AlreadyExists Agent already exists
blueman-applet 20.48.05 INFO     ShowConnected:50 enumerate_connections: Found 0 existing connections
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock object at 0x7f9c84150310>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DisconnectItems.DisconnectItems object at 0x7f9c84150a30>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.DiscvManager.DiscvManager object at 0x7f9c841509d0>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.NMPANSupport.NMPANSupport object at 0x7f9c84150b20>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.KillSwitch.KillSwitch object at 0x7f9c841508b0>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.AutoConnect.AutoConnect object at 0x7f9c84151bd0>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Networking.Networking object at 0x7f9c841507c0>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.StatusNotifierItem.StatusNotifierItem object at 0x7f9c84150160>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.StandardItems.StandardItems object at 0x7f9c84150280>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.TransferService.TransferService object at 0x7f9c84150250>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.NMDUNSupport.NMDUNSupport object at 0x7f9c841501f0>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.AuthAgent.AuthAgent object at 0x7f9c84151b70>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.RecentConns.RecentConns object at 0x7f9c841500d0>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.PowerManager.PowerManager object at 0x7f9c841519f0>
[... cut ...]
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.ShowConnected.ShowConnected object at 0x7f9c84151ed0>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <StatusIcon.StatusIcon object at 0x7f9c8926c9c0 (blueman+plugins+applet+StatusIcon+StatusIcon at 0x561264bdea60)>
blueman-applet 20.48.05 DEBUG    BasePlugin:63 _on_plugin_delete: Deleting plugin instance <blueman.plugins.applet.Menu.Menu object at 0x7f9c87e6dae0>
dell@DELL:~$ bluetoothctl list
Controller AC:D1:B8:CA:ED:CA DELL [default]
```

RubenFalcao14, 2024-02-20:
My system info :

```
System:
  Kernel: 5.15.0-94-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: Cinnamon 6.0.4
    tk: GTK 3.24.33 wm: muffin vt: 7 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Portable System: Dell product: Inspiron 3543 v: A08 serial: <superuser required> Chassis:
    type: 8 serial: <superuser required>
  Mobo: Dell model: 0YMC95 v: A00 serial: <superuser required> UEFI: Dell v: A08
    date: 05/13/2016
Battery:
  ID-1: BAT0 charge: 18.9 Wh (61.2%) condition: 30.9/32.6 Wh (95.0%) volts: 14.9 min: 14.8
    model: LGC-LGC3.0 DELL 49VTP27 type: Unknown serial: <filter> status: Discharging
CPU:
  Info: dual core model: Intel Core i3-5005U bits: 64 type: MT MCP smt: enabled arch: Broadwell
    rev: 4 cache: L1: 128 KiB L2: 512 KiB L3: 3 MiB
  Speed (MHz): avg: 1896 min/max: 500/1900 cores: 1: 1896 2: 1896 3: 1896 4: 1896
    bogomips: 15963
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel HD Graphics 5500 vendor: Dell driver: i915 v: kernel ports: active: eDP-1
    empty: HDMI-A-1 bus-ID: 00:02.0 chip-ID: 8086:1616 class-ID: 0300
  Device-2: Sunplus Innovation Integrated_Webcam_HD type: USB driver: uvcvideo bus-ID: 1-1.5:5
    chip-ID: 1bcf:2b8b class-ID: 0e02
  Display: x11 server: X.Org v: 1.21.1.4 driver: X: loaded: modesetting unloaded: fbdev,vesa
    gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1366x768 s-dpi: 96 s-size: 361x203mm (14.2x8.0") s-diag: 414mm (16.3")
  Monitor-1: eDP-1 model: Chi Mei Innolux res: 1366x768 hz: 60 dpi: 101
    size: 344x193mm (13.5x7.6") diag: 394mm (15.5") modes: 1366x768
  OpenGL: renderer: Mesa Intel HD Graphics 5500 (BDW GT2) v: 4.6 Mesa 23.2.1-1ubuntu3.1~22.04.2
    direct render: Yes
Audio:
  Device-1: Intel Broadwell-U Audio vendor: Dell driver: snd_hda_intel v: kernel bus-ID: 00:03.0
    chip-ID: 8086:160c class-ID: 0403
  Device-2: Intel Wildcat Point-LP High Definition Audio vendor: Dell driver: snd_hda_intel
    v: kernel bus-ID: 00:1b.0 chip-ID: 8086:9ca0 class-ID: 0403
  Sound Server-1: ALSA v: k5.15.0-94-generic running: yes
  Sound Server-2: PulseAudio v: 15.99.1 running: yes
  Sound Server-3: PipeWire v: 0.3.48 running: yes
Network:
  Device-1: Broadcom BCM43142 802.11b/g/n vendor: Dell Wireless 1704 802.11n + BT 4.0 driver: wl
    v: kernel pcie: speed: 2.5 GT/s lanes: 1 bus-ID: 06:00.0 chip-ID: 14e4:4365 class-ID: 0280
  IF: wlp6s0 state: up mac: <filter>
  Device-2: Realtek RTL810xE PCI Express Fast Ethernet vendor: Dell driver: r8169 v: kernel
    pcie: speed: 2.5 GT/s lanes: 1 port: 3000 bus-ID: 07:00.0 chip-ID: 10ec:8136 class-ID: 0200
  IF: enp7s0 state: down mac: <filter>
Bluetooth:
  Device-1: Broadcom BCM43142 Bluetooth 4.0 type: USB driver: btusb v: 0.8 bus-ID: 1-1.6:6
    chip-ID: 0a5c:21d7 class-ID: fe01 serial: <filter>
  Report: hciconfig ID: hci0 rfk-id: 3 state: down bt-service: enabled,running rfk-block:
    hardware: no software: yes address: <filter>
Drives:
  Local Storage: total: 465.76 GiB used: 27.03 GiB (5.8%)
  ID-1: /dev/sda vendor: Western Digital model: WD5000LPVX-75V0TT0 size: 465.76 GiB
    speed: 6.0 Gb/s type: HDD rpm: 5400 serial: <filter> rev: 1A01 scheme: GPT
Partition:
  ID-1: / size: 456.89 GiB used: 27.02 GiB (5.9%) fs: ext4 dev: /dev/sda2
  ID-2: /boot/efi size: 511 MiB used: 6.1 MiB (1.2%) fs: vfat dev: /dev/sda1
Swap:
  ID-1: swap-1 type: file size: 2 GiB used: 61 MiB (3.0%) priority: -2 file: /swapfile
USB:
  Hub-1: 1-0:1 info: Full speed or root hub ports: 2 rev: 2.0 speed: 480 Mb/s chip-ID: 1d6b:0002
    class-ID: 0900
  Hub-2: 1-1:2 info: Intel Integrated Hub ports: 8 rev: 2.0 speed: 480 Mb/s chip-ID: 8087:8001
    class-ID: 0900
  Device-1: 1-1.2:3 info: SiGma Micro Keyboard TRACER Gamma Ivory type: Keyboard,HID
    driver: hid-generic,usbhid interfaces: 2 rev: 1.1 speed: 1.5 Mb/s power: 98mA chip-ID: 1c4f:0002
    class-ID: 0300
[... 7 lines cut ...]
    driver: rtsx_usb,rtsx_usb_ms,rtsx_usb_sdmmc interfaces: 1 rev: 2.0 speed: 480 Mb/s power: 500mA
    chip-ID: 0bda:0129 class-ID: ff00 serial: <filter>
  Hub-3: 2-0:1 info: Hi-speed hub with single TT ports: 11 rev: 2.0 speed: 480 Mb/s
    chip-ID: 1d6b:0002 class-ID: 0900
  Hub-4: 3-0:1 info: Super-speed hub ports: 4 rev: 3.0 speed: 5 Gb/s chip-ID: 1d6b:0003
    class-ID: 0900
Sensors:
  System Temperatures: cpu: 52.0 C mobo: 47.0 C
  Fan Speeds (RPM): cpu: 0
Repos:
  Packages: apt: 2172
  No active apt repos in: /etc/apt/sources.list
  Active apt repos in: /etc/apt/sources.list.d/official-package-repositories.list
    1: deb http: //packages.linuxmint.com virginia main upstream import backport
    2: deb http: //archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
    3: deb http: //archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
    4: deb http: //archive.ubuntu.com/ubuntu jammy-backports main restricted universe multiverse
    5: deb http: //security.ubuntu.com/ubuntu/ jammy-security main restricted universe multiverse
  Active apt repos in: /etc/apt/sources.list.d/vscode.list
    1: deb [arch=amd64,arm64,armhf] http: //packages.microsoft.com/repos/code stable main
Info:
  Processes: 241 Uptime: 1h 14m wakeups: 2 Memory: 3.73 GiB used: 2.35 GiB (62.9%) Init: systemd
  v: 249 runlevel: 5 Compilers: gcc: 11.4.0 alt: 11/12 Client: Unknown python3.10 client
  inxi: 3.3.13
```

NelsonMungai1, 2024-03-14:
same issue I can't view my devices to connect them

closed 2024-05-21 by github-actions[bot]