blueman-project/blueman #2271: Blueman works fine, but offers error code when system boots.

1227mint, 2024-02-20:
Hello, I am new to Mint and having this issue. Every time I boot my system this Blueman error pops up however blueman is working fine. I saw a similar issue where someone deleted a specific file on re-install but as a new user to mint - I don't know how to do that? 

This started after I ran the disk out of space. Once I added plenty more space - this error persists. 

Thanks! 

g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 156, in _handle_method_call
    ok(method(*args))
  File "/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Network.py", line 56, in _reload_network
    nc = NetConf.get_default()
  File "/usr/lib/python3/dist-packages/blueman/main/NetConf.py", line 313, in get_default
    obj: "NetConf" = pickle.load(f)
EOFError: Ran out of input
 (0)

JokeyTheSmurf, 2025-03-27:
> What blueman is trying to do is read `/var/lib/blueman/network.state` and failing. Deleting it will most likely fix this. But can you attach the file here before you delete it as I want to have a quick look at it.

I saw the same error today after accidentally filling `/` to the brim, with 0 bytes of free space left.
Deleting `network.state` solved the issue. Thanks :)

infirit, 2025-03-28:
We moved away from using `network.state` in blueman 2.4. Closing.

closed 2025-03-28 by infirit