blueman-project/blueman #2282: `blueman-applet` won't start on Wayland

alba4k, 2024-02-28:
blueman: 2.3.5
BlueZ: 5.72
Distribution: Arch Linux
Desktop environment: Hyprland

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

```
blueman-applet 22.58.44 WARNING  PluginManager:94 load_plugin: Failed to start plugin GameControllerWakelock: Only X11 platform is supported
blueman-applet 22.58.44 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 22.58.44 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
```

infirit, 2024-02-28:
You posted a non fatal warning. Post a proper blueman-applet debug log as described https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging-blueman

alba4k, 2024-03-01:
Well idk what changed but it's starting now

Still no icon in the tray tho

`blueman-tray --loglevel debug`
```
blueman-tray 18.50.52 DEBUG    Tray:57 _on_name_vanished: Applet shutdown or not available at startup
```

alba4k, 2024-03-01:
To be clear, blueman does work and display notifications

Also, running that did not help

The only problem is that the tray icon fails to display (at least, on hyprland)

cschramm, 2024-03-01:
So the blueman-applet process is running and working but running blueman-tray gives `Tray:57 _on_name_vanished: Applet shutdown or not available at startup` and dies?

alba4k, 2024-03-01:
If I try to run `blueman-tray` while `blueman-applet --loglevel debug` is running, the latter will print `(blueman-tray:1543409): Gtk-CRITICAL **: 22:46:47.057: gtk_widget_get_scale_factor: assertion 'GTK_IS_WIDGET (widget)' failed` while the former outputs nothing and returns 0

cschramm, 2024-03-02:
For some reason `blueman-tray` cannot find the applet on the D-Bus session bus. Do you see `org.blueman.Applet` e.g. in `dbus-send --print-reply --dest=org.freedesktop.DBus  /org/freedesktop/DBus org.freedesktop.DBus.ListNames`?

alba4k, 2024-03-02:
Yes, I see both `org.blueman.Applet` and `org.blueman.Tray` (after starting only `blueman-applet`, manually at least). I don't see any of those after killing that process

infirit, 2024-03-02:
To be sure, blueman-applet is the core of blueman and it's supposed to always be running. The applet is what (re)starts blueman-tray (the tray icon). If you are killing it all the time stop that.

infirit, 2024-03-02:
Installed hyprlan and waybar. Then setup some basic environment variables for QT, Gtk, XDG_* and a couple others. Added exec-once for waybar and below that exec-once blueman-applet. It worked flawlessly.

Sound to me like you either are completely missing a place for tray icons, what eg waybar provides or your setup is broken. It's not an issue in blueman.

cschramm, 2024-03-03:
> Yes, I see both `org.blueman.Applet` and `org.blueman.Tray`

blueman-tray is already running then.

alba4k, 2024-03-03:
> blueman-tray is already running then.

Only if I start the applet. nothing is shown if I dont

Yes I have a system tray set up. Yes it works for everything else. Yes I see the tray icon on polybar on i3

cschramm, 2024-03-03:
> I have a system tray set up

... which is ... waybar? Does it give any warnings or related debug output when you start blueman-applet and blueman-tray?

alba4k, 2024-03-04:
Yes, it is waybar.

This is everything I get from `blueman-applet

<details>

<summary>output</summary>

```
blueman-applet 22.31.24 INFO     PluginManager:87 load_plugin: ['DiscvManager', 'Menu', 'NMDUNSupport', 'NMPANSupport', 'KillSwitch', 'RecentConns', 'Networking', 'DhcpClient', 'TransferService', 'StatusNotifierItem', 'NetUsage', 'ConnectionNotifier', 'ShowConnected', 'GameControllerWakelock', 'PowerManager', 'SerialManager', 'ExitItem', 'AuthAgent', 'AutoConnect', 'DisconnectItems', 'DBusService', 'StandardItems', 'PPPSupport', 'StatusIcon']
blueman-applet 22.31.24 WARNING  PluginManager:94 load_plugin: Failed to start plugin GameControllerWakelock: Only X11 platform is supported
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 22.31.24 DEBUG    Base:60 do_g_properties_changed: /org/bluez/hci0 {'Address': '38:7A:0E:DB:63:BF', 'AddressType': 'public', 'Name': 'dell-xps', 'Alias': 'dell-xps', 'Class': 7078156, 'Powered': True, 'PowerState': 'on', 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': False, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '0000184e-0000-1000-8000-00805f9b34fb', '00001845-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001844-0000-1000-8000-00805f9b34fb', '0000184d-0000-1000-8000-00805f9b34fb', '00001843-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '0000180a-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '0000111f-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001850-0000-1000-8000-00805f9b34fb', '0000184f-0000-1000-8000-00805f9b34fb', '0000111e-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0548', 'Roles': ['central', 'peripheral', 'central-peripheral'], 'ExperimentalFeatures': ['671b10b5-42c0-4696-9227-eb28d1b049d6', '330859bc-7506-492d-9370-9a6f0614037f', '15c0a148-c273-11ea-b3de-0242ac130004', 'a6695ace-ee7f-4fb9-881a-5fac66c629af', '6fbaf188-05e0-496a-9885-d6ddfdb4e03e'], 'Manufacturer': 2, 'Version': 12}
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 22.31.24 INFO     Networking:35 load_nap_settings: Loading NAP settings
blueman-applet 22.31.24 WARNING  PluginManager:151 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 22.31.24 WARNING  PluginManager:151 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ConnectionNotifier.ConnectionNotifier'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.AutoConnect.AutoConnect'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.DisconnectItems.DisconnectItems'>
blueman-applet 22.31.24 INFO     PluginManager:160 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 22.31.24 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 22.31.24 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 22.31.24 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 22.31.24 INFO     KillSwitch:85 _on_connman_vanished: net.connman vanished
blueman-applet 22.31.24 INFO     TransferService:244 _on_dbus_name_vanished: org.bluez.obex not running or was stopped
[... cut ...]
blueman-applet 22.31.25 INFO     PowerManager:125 check     : callbacks done
blueman-applet 22.31.25 INFO     PowerManager:96 set_adapter_state: True
blueman-applet 22.31.25 INFO     PowerManager:179 update_power_state: off False | foff False | on True | current state True | new state True
```

</details>

Additionally, this is what I get every time I try to manually start `blueman-tray`, in the applet's output

```
(blueman-tray:10283): Gtk-CRITICAL **: 22:31:42.539: gtk_widget_get_scale_factor: assertion 'GTK_IS_WIDGET (widget)' failed
```

Enigma1309, 2024-03-11:
I'm getting the exact same issue, after  killing and relaunching waybar, due to this my other systray icons are also getting hidden

`(blueman-tray:62204): Gtk-CRITICAL **: 10:54:48.544: gtk_widget_get_scale_factor: assertion 'GTK_IS_WIDGET (widget)' failed`

cschramm, 2024-03-11:
Oh, I did not get that the GTK assertion message comes from `blueman-tray`, not `blueman-applet`. That makes a lot more sense, then. :sweat_smile:

Just for better understanding of the linked issue, @Enigma1309: Did this actually start with that hyperdots update? You first stated that all icons are affected, not just blueman-tray, and later stated that it would be an issue with blueman-tray. So which one is correct or are you saying that blueman-tray breaks the other icons as well?

Enigma1309, 2024-03-11:
> blueman-tray breaks the other icons as well?

yup that's what i think so, since you can see that when relaunching waybar the only thing that's consuming a lot of processing power is blueman tray 

Just did some testing and found this,
[image]
i removed  blueman and relaunching the waybar  shows the network-manager icon, it's probably a blueman issue

just let me reinstall and reboot now to see if the problem persists

cschramm, 2024-03-11:
Well, waybar definitely lacks separation there then, but anyway...

I think I get it. You're running into the GTK indicator implementation and getting exactly what's expected when the environment is not properly set up for the graphical session.

On X, with the StatusNotifier plugin disabled (I guess it does not work with waybar, so that blueman-tray falls back to the GTK status icon anyway), blueman-applet already running and *no* blueman-tray running (just `killall blueman-tray` when blueman-applet started one), `DISPLAY= blueman-tray --loglevel debug` gives me this:

```
blueman-tray 12.17.32 DEBUG    Tray:36 _on_name_appeared: Applet started on name org.blueman.Applet, showing indicator

(blueman-tray:257434): Gtk-CRITICAL **: 12:17:32.526: gtk_widget_get_scale_factor: assertion 'GTK_IS_WIDGET (widget)' failed
blueman-tray 12.17.32 INFO     Tray:46 _on_name_appeared: Using indicator "GtkStatusIcon"

(blueman-tray:257434): Gtk-CRITICAL **: 12:17:32.536: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(blueman-tray:257434): Gtk-CRITICAL **: 12:17:32.536: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(blueman-tray:257434): Gtk-CRITICAL **: 12:17:32.536: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
```

and `SIGSEGV`s.

So... whatever starts blueman-applet for you, which in terms starts blueman-tray, does not provide a proper environment for GTK to work. On X it means no proper `DISPLAY`, but I cannot recall if it's `DISPLAY` as well on Wayland.

Enigma1309, 2024-03-11:
[image]

i think display configuration is good from my side,

anyways...
I've reinstalled `blueman` after clearing pacman cache, rebooted the machine( thrice just to be sure ) and now i think it works
idk what's the cause of this issue but now it has started working.
thanks for putting your time in this
Thank You <3

cschramm, 2024-03-11:
> I've reinstalled `blueman` after clearing pacman cache

That's definitely not something that should fix such an issue. Did you happen to install a different version / revision?

Enigma1309, 2024-03-11:
nope just did `yay -S blueman`

Enigma1309, 2024-03-11:
here are logs of the previous session if you're interested
[image]

infirit, 2024-03-11:
> So... whatever starts blueman-applet for you, which in terms starts blueman-tray, does not provide a proper environment for GTK to work. On X it means no proper `DISPLAY`, but I cannot recall if it's `DISPLAY` as well on Wayland.

If blueman is using the Gtk.StatusIcon instead of the StatusNotifier then it's expected it breaks as it relies on X11. On properly configured environments we can check for `XDG_SESSION_TYPE` environment variable. But as we regularly see badly configured environments its probably safer to import `GdkX11` and if it fails we know it's definitely not X11. 

Perhaps we should promote the StatusNotifier plugin as the default and disable the deprecated Gtk.StatusIcon.

cschramm, 2024-03-11:
> If blueman is using the Gtk.StatusIcon instead of the StatusNotifier then it's expected it breaks as it relies on X11.

Oh, true, GTK's Wayland backend does not support it. If that's the only problem, I would expect it to fail silently, though.

> Perhaps we should promote the StatusNotifier plugin as the default and disable the deprecated Gtk.StatusIcon.

I'll have to double check, but it should be the default already, no?

I do have an idea what could go wrong on restarts even if of the StatusNotifierItem is enabled and used. Will check.

lvindotexe, 2024-03-15:
currently only running blueman applet
```
blueman-tray: no process found
blueman-tray 17.08.59 DEBUG    Tray:35 _on_name_appeared: Applet started on name org.blueman.Applet, showing indicator
blueman-tray 17.08.59 INFO     Tray:45 _on_name_appeared: Using indicator "StatusNotifierItem"
```

alba4k, 2024-05-15:
that makes no sense

closing bugs just because they're not recieving work?

alba4k, 2024-05-21:
disabling and re-enabling the plugin fixed that

closed 2024-05-21 by alba4k

johnwmail, 2024-06-01:
> disabling and re-enabling the plugin fixed that

How to disable / enable the plugin? thanks.

LCarmi, 2024-06-20:
I've had the same issue in the past week; can confirm that disabling and re-enabling the `StatusIcon` and `StatusNotifierItem` fixed the problem for me.

@johnwmail you can find the plugins in the `blueman-manager` app at `View > Plugins`

tom-gora, 2024-08-22:
As a workaround I am running this script on startup:
`#!/bin/bash

# Initial attempt to start the blueman-applet
blueman-applet &
# Sleep for 5 seconds to make sure attempt executed
sleep 5
# Do the first check and kill the first instance that fails to draw icon
if pgrep -x "blueman-applet" >/dev/null; then
	pkill -x "blueman-applet"
	sleep 2
fi
# tragain
blueman-applet &
# Check if ok and if ok exit before the loop
if pgrep -x "blueman-applet" >/dev/null; then
	exit 0
fi

# Init the counter to not loop indefinitely but do max of 10 attrempts
counter=0
# Loop to try and start the blueman-applet
while [ $counter -lt 10 ]; do
	# Try to start blueman-applet
	blueman-applet &
	# Sleep for 2 seconds before the next check...
	sleep 2
	# Check if success
	if pgrep -x "blueman-applet" >/dev/null; then
		exit 0
	# ... and prompt if attempt failed.
	else
		notify-send -u normal "Attempt $((counter + 1)) to start bluetooth tray applet failed."
	fi
	((counter++))
done

# Final check and if all fails then prompt
if ! pgrep -x "blueman-applet" >/dev/null; then
	notify-send -u critical "Couldn't start bluetooth tray applet!"
fi
`
It basically kills the first instance of the applet that for whatever reason fails to draw tray icon on startup every time and attempts a start the package again, which is seemed to work every time for me when done maunally, so I scripted it and it seems to force it to work...

cschramm, 2024-10-16:
If anyone's interested in getting this fixed, I'd still be interested in `dconf read /org/blueman/general/plugin-list` from an affected setup. The `--loglevel debug` output of an automatically started instance that shows the misbehavior would also be helpful (possibly the standard error output could help as well already).

edoardomussi, 2024-10-29:
`dconf read /org/blueman/general/plugin-list` has output `['StatusIcon', 'ShowConnected', 'StatusNotifierItem']`
@cschramm

xulongwu4, 2025-01-10:
I don't know if my issue is the same as the one mentioned here. I see that blueman-applet and blueman-tray are running on my side but there is no blueman icon on the waybar's system tray. However, launching blueman-manager results in a segfault, the same symptom as mentioned in https://github.com/dslm4515/BMLFS/issues/24#issuecomment-1369708085

cschramm, 2025-01-11:
You disabled the `StatusIcon` plugin. It is expected behavior that no icon shows up then. You could find the plugins dialog in blueman-manager's "View" menu to re-enable it. .. that is, if blueman-manager would work, of course. :see_no_evil: In any case, you can `dconf reset /org/blueman/general/plugin-list`.

The `bad_cast` seems to be https://github.com/Alexays/Waybar/issues/3597 (which only affects waybar 0.11.0 and while there is no patch release, there is a fix in both, upstream and (most?) distribution packages).

xulongwu4, 2025-01-13:
I reinstalled my system and blueman-manager works on my wayland environment now, and I can see the blueman icon on waybar by including the waybar tray fix mentioned above.

WDMdanila, 2025-03-12:
For me disabling `StatusNotifierItem` shows tray icon and enabling it hides it again. Not Wayland though but xfce4