blueman-project/blueman #2297: Better fallback incoming folder selection

drws, 2024-03-16:
blueman: v2.3.5
BlueZ: 5.73
Distribution: Arch
Desktop environment: Xfce

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

If the configured incoming transfer folder is not available, Blueman falls back to home folder, notifying user:

> **Configured directory for incoming files does not exist**
> Please make sure that directory `/home/user/Incoming` exists or configure it with blueman-services. Until then the default `/home/user` will be used

`$HOME` is not the best first fallback choice. Much more appropriate would be `$XDG_DOWNLOAD_DIR` or `$XDG_PUBLICSHARE_DIR`. Only if the system doesn't have XDG configured then Blueman could fallback to the home folder.

infirit, 2024-03-16:
I disagree

drws, 2024-03-18:
Ok. Is there any other option sensible to you than the root of the home directory structure?

cschramm, 2024-03-18:
`$XDG_PUBLICSHARE_DIR` definitely does not seem like a good idea. Also, as far as I know those variables are not specified in any way. They just come from the `xdg-user-dirs` tools. The only suitable variable from the XDG Base Directory Specification would be `$XDG_DATA_HOME`. It defaults to ` $HOME/.local/share`, meaning that it's "hidden", thus doesn't seem appropriate for incoming files either.

`$XDG_DOWNLOAD_DIR` might be reasonable, if set. :shrug: @infirit, what are your points against it?

infirit, 2024-03-18:
Zero reason was given why $HOME is not the best fallback. How is using the download dir any better or worse?

drws, 2024-03-20:
So as not to litter the root of an important directory structure where user expects full control of their files. Poorly-ported commercial software does that, FOSS mostly tries to avoid it. And even then files are usually put in a custom subfolder. So tidiness and software hygiene is one.

Also a very great majority of userspace applications categorize their files by default in subfolders so user actually expects files in an appropriate standard subfolder. `$XDG_DOWNLOAD_DIR` seems like the best choice.

Furthermore, security. To try to limit any potential attacks. There are some important config files and scripts in root of home. It doesn't seem the best folder to push 3rd party data files to. Also a bug could cause data flood or corruption. In all these cases there's simply less damage and damage control is easier if files are categorized/contained at least in one of the general data subfolders.

Even further, convenience. A sensible falback=default selection means there's one less path to worry about when setting up a system.