blueman-project/blueman #2312: Device not shown (text doesn't appear) in devices list but it work

nebulosa2007, 2024-04-01:
blueman: 2.4
BlueZ: 5.73-4
Distribution: Arch Linux
Desktop environment: Niri (wayland)

* [ x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

After upgrading to version 2.4, my Bluetooth speaker is not listed, but I can interact with it. When left-clicking, a drop-down menu appears (e.g. "Rename" is selected ):

[image]

[Log from the terminal](https://0x0.st/Xzf0.txt) (it adds more lines on hovering over the selected line). The `Trust` item also doesn't work (last 15 lines)

cschramm, 2024-04-01:
The crucial issue is `Icon 'audio-card' not present in theme Adwaita`. As `audio-card` definitely is part of the 
Adwaita Icon Theme, you seem to have an issue with your installed and active theme. In https://github.com/blueman-project/blueman/issues/2101#issuecomment-1663787627 just switching themes forth and back did the trick to fix it. :shrug:

infirit, 2024-04-02:
The fallback I added for this is broken. Blueman now skips making an IconInfo object which would return None instead of raising GEerror.

nebulosa2007, 2024-04-02:
> In [#2101 (comment)](https://github.com/blueman-project/blueman/issues/2101#issuecomment-1663787627) just switching themes forth and back did the trick to fix it. 🤷

I don't know how it can do this, I haven't gnome environment or Gnome setting apps..

Btw, after installing [gnome-icon-theme package](https://aur.archlinux.org/packages/gnome-icon-theme) all the errors disappeared. I think this is an obsolete package in Arch, so I need other thoughts, If any.

`adwaita-icon-theme` package is already installed as a dependency of `gtk3` and `gtk4`

[image]

cschramm, 2024-04-02:
> > In [#2101 (comment)](https://github.com/blueman-project/blueman/issues/2101#issuecomment-1663787627) just switching themes forth and back did the trick to fix it. 🤷
> 
> I don't know how it can do this, I haven't gnome environment or Gnome setting apps..

If there's no specific support for it, editing `$HOME/.config/gtk-3.0/settings.ini` should do the trick, possibly combined with some kind of reload, but read on...

> Btw, after installing [gnome-icon-theme package](https://aur.archlinux.org/packages/gnome-icon-theme) all the errors disappeared. I think this is an obsolete package in Arch, so I need other thoughts, If any.

They came back when uninstalling, I suppose?

> `adwaita-icon-theme` package is already installed as a dependency of `gtk3` and `gtk4`

Exactly. It's a vital part of GTK as it ships defaults that all themes fall back to. That's why there is no good reason for `audio-card` to be unavailable.

My first guess would be some kind of broken cache. I'd try `gtk-update-icon-cache` (and probably switching icon themes forth and back has exactly that effect).

Another option could be a broken icon theme, that actually breaks the fallbacks. Not sure if that's possible.

nebulosa2007, 2024-04-02:
> If there's no specific support for it, editing `$HOME/.config/gtk-3.0/settings.ini` should do the trick, possibly combined with some kind of reload, but read on...

I don't have files ..../gtk-3.0/settings.ini or ..../gtk-4.0/settings.ini

> They came back when uninstalling, I suppose?

Yes, exactly.

> My first guess would be some kind of broken cache. I'd try `gtk-update-icon-cache` (and probably switching icon themes forth and back has exactly that effect).
> Another option could be a broken icon theme, that actually breaks the fallbacks. Not sure if that's possible.

```console
$ ls -1d /usr/share/icons/* | tee /dev/tty | xargs -I {} sudo gtk-update-icon-cache -f {}
/usr/share/icons/Adwaita/
/usr/share/icons/default/
/usr/share/icons/hicolor/
gtk-update-icon-cache: Cache file created successfully.
gtk-update-icon-cache: Cache file created successfully.
```
Didn't help again.

So it's just a distribution issue on Arch?

cschramm, 2024-04-02:
A simple test for a `python` shell:

```python
import gi
gi.require_version("Gtk", "3.0")
from gi.repository import Gtk
Gtk.IconTheme.get_default().load_surface("audio-card", 48, 1, None, Gtk.IconLookupFlags.FORCE_SIZE)
```

This is basically what blueman does, except that it gets the scale factor from GTK, so that it could be something else than 1 (typically 2) and it passes a window.

I expect this to always work as Adwaita ships a scalable `audio-card` icon with specified sizes between 8 and 512.

Just to double check: `/usr/share/icons/default/` simply has an `index.theme` with the following, correct?

```
[Icon Theme]
Inherits=Adwaita
```

nebulosa2007, 2024-04-02:
Yes:
```console 
$ cat /usr/share/icons/default/index.theme 
[Icon Theme]
Inherits=Adwaita
```

And test output is:
```console
$ python test.py 
Traceback (most recent call last):
  File "/home/nebulosa/test.py", line 4, in <module>
    Gtk.IconTheme.get_default().load_surface("audio-card", 48, 1, None, Gtk.IconLookupFlags.FORCE_SIZE)
gi.repository.GLib.GError: gtk-icon-theme-error-quark: Icon 'audio-card' not present in theme Adwaita (0)
```

cschramm, 2024-04-02:
Might be best to file a bug with the adwaita-icon-theme Arch package to get behind that. From my understanding, there is no way for it to not provide the icon. The path is basically: `Directories` in `/usr/share/icons/Adwaita/index.theme` containing `symbolic/devices` and a respective section:

```
[symbolic/devices]
Context=Devices
Size=16
MinSize=8
MaxSize=512
Type=Scalable
```

=> `/usr/share/icons/Adwaita/symbolic/devices/audio-card-symbolic.svg` gets used for sizes of 8 to 512.

nebulosa2007, 2024-04-02:
```console
$ grep -A5 "\[symbolic/devices\]" /usr/share/icons/Adwaita/index.theme
[symbolic/devices]
Context=Devices
Size=16
MinSize=8
MaxSize=512
Type=Scalable

$ file /usr/share/icons/Adwaita/symbolic/devices/audio-card-symbolic.svg
/usr/share/icons/Adwaita/symbolic/devices/audio-card-symbolic.svg: SVG Scalable Vector Graphics image
```

I'll make an issue on a Archlinux bugtracker then.

infirit, 2024-04-02:
> The fallback I added for this is broken. Blueman now skips making an IconInfo object which would return None instead of raising GEerror.

#2316 fixes the fallback for broken icon themes.

nebulosa2007, 2024-04-03:
New info! After link file (or copy) and update icons cache all errors are gone:
```console
$ sudo ln -s /usr/share/icons/Adwaita/symbolic/devices/audio-card-symbolic.svg /usr/share/icons/Adwaita/symbolic/devices/audio-card.svg 

$ ls -1d /usr/share/icons/* | tee /dev/tty | xargs -I {} sudo gtk-update-icon-cache -f {}
/usr/share/icons/Adwaita/
/usr/share/icons/default/
/usr/share/icons/hicolor/
gtk-update-icon-cache: Cache file created successfully.
gtk-update-icon-cache: Cache file created successfully.
```
[image]
Btw, this icon is different than above. 

I think that `blueman` looking for exactly `audio-card.*` but not `audio-card<whatever>.*` file and take the biggest one as possible. I found icon above in `/usr/share/icons/gnome/48x48/devices/audio-card.png`

cschramm, 2024-04-03:
I'm totally confused, but, yes, the symbolic version is the only one in the package and it does not seem to get used unless requested explicitly (by name or with `Gtk.IconLookupFlags.FORCE_SYMBOLIC`). It looks like adwaita-icon-theme 44 dropped non-symbolic icons, not even reflecting that in the changelog.

infirit, 2024-04-03:
There are a bunch of issues on the adwaita icon theme gitlab about full corour icons that got dropped the 44 release. The issues I checked were fixed by re-adding them.

cschramm, 2024-04-03:
Only very few of the icons I listed in https://github.com/blueman-project/blueman/issues/976#issuecomment-450735133 are still available as non-symbolic, namely `computer`, `image-missing`, `input-*`, `network-workgroup`, `phone` and `printer`. Wtf... :confused:

infirit, 2024-04-03:
I took your list and checked which one I had on my system.

Missing
- audio-card
- camera-photo
- camera-video
- modem
- network-wireless

In scalable directory
- computer 
- input-gaming
- input-keyboard
- input-mouse
- input-tablet
- phone
- printer

I don't use Adwaita as icon theme so I never noticed them missing.

cschramm, 2024-04-05:
That's just one part of the list, namely names provided by BlueZ. Of all the other icons that we use `image-missing` and `network-workgroup` are the only ones still available.

cschramm, 2024-04-05:
Still unsure where to go from here. One obvious solution would be to always use symbolic icons for everything...

cschramm, 2024-04-05:
Oh, forget what I wrote. We're already using symbolic icons in many places anyway. Is this actually just an issue with BlueZ-provided device icons? :thinking: Would be easy to force symbolic ones there of course.

nebulosa2007, 2024-04-05:
Let's try!
```console 
# Check files:
$ file /usr/share/icons/Adwaita/symbolic/devices/audio-card-symbolic.svg
/usr/share/icons/Adwaita/symbolic/devices/audio-card-symbolic.svg: SVG Scalable Vector Graphics image
$ file  /usr/share/icons/Adwaita/symbolic/devices/audio-card.svg
/usr/share/icons/Adwaita/symbolic/devices/audio-card.svg:          cannot open `/usr/share/icons/Adwaita/symbolic/devices/audio-card.svg' (No such file or directory)
```

Built and installed packages with patches, here are the results:
2.4 + #2316: 
[image]

2.4 + #2330:
[image]

2.4 + #2316 + #2330:
[image]

cschramm, 2024-04-05:
> But those icons are still broken: ![image](https://private-user-images.githubusercontent.com/85841412/320049493-e6e5284a-4df5-4e79-b27d-2a471e1b9641.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTIzNTEzOTEsIm5iZiI6MTcxMjM1MTA5MSwicGF0aCI6Ii84NTg0MTQxMi8zMjAwNDk0OTMtZTZlNTI4NGEtNGRmNS00ZTc5LWIyN2QtMmE0NzFlMWI5NjQxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MDUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDA1VDIxMDQ1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWI1YzRmNDJmODEwMTI0ODMyYThkMzE0MTBkMjQxYTMwZjI1NDBmYWNhMTRlOTExZTA1NTUwYTk3YWIyN2UxNWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.3NgtK_VVxcH7VsbuAFHgCY5jZMrfLknDnDdXY6v84UU)

I don't get what you mean there.

> Edit: After reboot (?) icons got back. So for me, it seems, that issue can be closed. All icons got back

You're not saying you have device icons without any patches now, are you? :sweat_smile:

nebulosa2007, 2024-04-06:
> I don't get what you mean there.

Never mind. Icons (orange and blue) appears when device is connected. Didn't notice it before. 

> You're not saying you have device icons without any patches now, are you? 😅

Nope. Device icons appear when I apply the #2330 patch (with or without patch #2316) and build the package afterwards. For now I will wait for upstream updates and use the patched version.

P.S.
Not only `blueman` has this issue, I checked also `pavucontrol` and [found ](https://gitlab.freedesktop.org/pulseaudio/pavucontrol/-/issues/153) the [rabbit hole](https://gitlab.freedesktop.org/xdg/xdg-specs/-/issues/128)

infirit, 2024-04-06:
Typical gnome behaviour. Looks like we need resurrect our old icons or find a new set.

cschramm, 2024-04-08:
I fail to find a proper primary source, but at least https://gitlab.gnome.org/GNOME/adwaita-icon-theme/-/issues/242 makes it very clear that

> a-i-t is no longer some icon and cursor pack for everyone to use and apps should bundle their own icons

Unfortunately that effectively means that there just is no "icon pack for everyone to use", it seems. Wow, what a mess... :unamused: 

Well, we can just choose an icon theme that we like and that fits our license (fingers crossed :crossed_fingers: :sweat_smile:) and bundle their icons. Most desktops will override them anyway and maintainers could strip them from their blueman packages (I do not really see how they set their packages up then, though, as they'd have to scan all their distribution's icon themes for compatibility to indicate that any of the compatible ones is a dependency of blueman).

To put it more precisely: Wow, what a ridiculous mess for the whole Linux desktop!

infirit, 2024-04-08:
We can add additional search paths to the Gtk.IconTheme and point it to fallback icons we provide in some directory we control. Something like `/usr/share/blueman/icons`. I kinda hope distributions will drop Adwaita as the default and use something that provides full color icons following the icon naming spec.

nebulosa2007, 2024-04-09:
> In the meantime, make sure that your packages depend on proper icons themes.
Arch Linux has no default icon themes, only those provided by program developers. That's how I discovered this bug. If it will be just `symbolic` icons that's will be fine I guess.

infirit, 2024-04-09:
They have to and do. The content of the file you checked before (below) can't inherit from Adwaita anymore. So whatever this is set to has to provide a full icon theme that applications should be able to rely on.

```
$ cat /usr/share/icons/default/index.theme 
[Icon Theme]
Inherits=Adwaita
```

infirit, 2024-05-04:
By now budgie, cinnamon and plasma have blacklisted adwaita in their system settings ui as it is no longer an fdo compatible icon theme.

My suggestion is we also enforce at lesst one compatibe theme at build time. Breeze can be added to the list as it has everything we need.

cschramm, 2024-05-10:
> Breeze can be added to the list as it has everything we need.

Does it? Querying just a few icon names from our code, it does not seem to have e.g. `application-x-addon-symbolic`, `modem-symbolic`, or `gnome-power-manager-symbolic` and we cannot assume a fallback on Adwaita for them, can we?

Papirus looks pretty complete from a quick glance.

Even worse than finding compatible themes, MATE's and Linux Mint's seem incompatible as well, they e.g. lack `application-x-addon-symbolic`.

clefebvre, 2024-05-10:
Hi @cschramm and @infirit,

Are you on Matrix or IRC? I'd love to help and get your feedback on this. It's affecting a lot of people and projects right now, DEs, distros, apps :)

infirit, 2024-05-10:
I'm on neither @clefebvre. The best place to get the latest on this is this issue created by Nate Graham from KDE, https://gitlab.gnome.org/GNOME/adwaita-icon-theme/-/issues/288.

Specifically [this comment](https://gitlab.gnome.org/GNOME/adwaita-icon-theme/-/issues/288#note_2101602) from the AIT maintainer is a reasonable solution.

cschramm, 2024-05-11:
I don't find e.g. `modem` in Tango, but https://gitlab.gnome.org/GNOME/adwaita-icon-theme/-/merge_requests/70 looks promising. Ideally AdwaitaLegacy would have scalables, but at least it should make Adwaita provide everything that we use again.

clefebvre, 2024-05-11:
Tango isn't maintained anymore :(

There's also an issue coming from the fact that Adwaita provides symbolics and fullcolors come from GNOME. The two-level inheritance leads to Adwaita symbolics taking priority over inherited Fullcolor.

clefebvre, 2024-05-11:
https://gitlab.gnome.org/jimmac/adwaita-icon-theme-legacy/ looks good indeed. I'll try and talk with Jakub about the symbolic inheritance.

The situation is definitely going to get fixed somehow and no matter what, individual apps won't need to adapt. This is something toolkits/DEs/distros need to guarantee to app developers.

nebulosa2007, 2024-05-30:
Seems that Issue is resolved - now all icons in apps is showed. If not, package maintainers should add [adwaita-icon-theme-legacy](https://gitlab.gnome.org/GNOME/adwaita-icon-theme-legacy) dependency

Now blueman looks like this - https://github.com/blueman-project/blueman/issues/2312#issuecomment-2031711758

closed 2024-05-30 by nebulosa2007

cschramm, 2024-05-30:
Awesome! The recently released Adwaita Icon Theme 46.2 added the inheritance from Adwaita Icon Theme legacy, so anybody experiencing this just needs to update to get icons back. :tada: