blueman-project/blueman #2317: Annoying battery notifications

JPenuchot, 2024-04-02:
blueman: 2.4
BlueZ: 5.73
Distribution: Arch Linux
Desktop environment: Sway

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi,

I'm getting a notification every time my earbuds' batteries drop by 1%. It is getting pretty annoying, how do I disable that?

Regards,
Jules

cschramm, 2024-04-02:
Right-click, Plugins, ConnectionNotifier

ouuan, 2024-04-07:
> Right-click, Plugins, ConnectionNotifier

I would like to see the battery status only once when the device is first connected, but not on subsequent updates (drop by 1%).

cschramm, 2024-04-08:
To be honest I haven't seen a device that provides battery updates with a 1 % granularity. We can make notifications configurable, of course, but could it be a reasonable tradeoff to only show new values if they are multiples of 10 % or would that still annoy you?

ouuan, 2024-04-08:
> We can make notifications configurable, of course, but could it be a reasonable tradeoff to only show new values if they are multiples of 10 % or would that still annoy you?

10% only sounds acceptable to me.

4eUeP, 2024-04-08:
I also find this annoying, especially since my keyboard's battery is "unstable". So I add an option to turn off battery updates: https://github.com/4eUeP/blueman/commit/5a9ac2259c719681178acf1b3062cafabdecaeef

> We can make notifications configurable
> could it be a reasonable tradeoff to only show new values if they are multiples of 10 %

How about showing notifications only when the battery level is lower than a value(e.g., 10%)?

Proposal: a new option, e.g., `min-battery-level: Int`. Notifications will be shown only if the current battery level is below the setting. `0` means disable battery-updates notifications.

JPenuchot, 2024-04-09:
> To be honest I haven't seen a device that provides battery updates with a 1 % granularity.

The Galaxy Buds2 Pro do that unfortunately :/

kuraga, 2024-04-09:
> I'm getting a notification every time my earbuds' batteries drop by 1%. It is getting pretty annoying, how do I disable that?

> Right-click, Plugins, ConnectionNotifier

Stop. How to **enable** them? See battery level on connection only.

cschramm, 2024-04-10:
Not sure about a minimum battery level. The purpose of the notifications is (or was) to inform about the current level. Most of that should be rather up to a power management tool and in many cases they probably do handle it by now (via UPower).

A "running low" logic in blueman seems a bit pointless in light of power management tools and an arbitrary threshold does not sound like a good idea, e.g. mice typically provide distinct levels like 100 %, 80 %, 50 % and 20 % for which a 10 % threshold would just never trigger.

The applet plugin for battery level notifications used to be separate and I'd say the main purpose was to show the level on connection. We merged it with `ConnectionNotifier` to show a single notification for the connection that *included* the battery level. The "depletion" notifications are more or less a leftover and we could also drop them completely. :shrug:

askmeaboutlo0m, 2024-04-10:
I have devices that can't decide what their battery level is supposed to be, they flicker back and forth between 100% and 90% several times as their battery depletes. So a change to the threshold to 10% would not fix the notification annoyance.

Turning off the ConnectionNotifier plugin altogether is fine by me, just mentioning this in case it's useful to help with the solution.

cschramm, 2024-04-12:
Well, ok, I don't see us going down that rabbit hole. So, as we do not want to get into power management applet business, I suggest that show the battery level on connection only and do not process any updates.

mar565, 2024-04-27:
> I also find this annoying, especially since my keyboard's battery is "unstable". So I add an option to turn off battery updates: [4eUeP@5a9ac22](https://github.com/4eUeP/blueman/commit/5a9ac2259c719681178acf1b3062cafabdecaeef)

How can I change this setting? Is there a config somewhere? Also the threshold idea seems the way to go.

ouuan, 2024-04-27:
> How can I change this setting? Is there a config somewhere?

There is a link to the fork, so the config is in that fork and is not merged here. You need to build the fork if you want to use it.

> Also the threshold idea seems the way to go.

The conversations above have concluded that the threshold idea is not the way to go. Instead, we will "show the battery level on connection only and do not process any updates". Please read them and give your argument if you disagree.

cschramm, 2024-05-10:
Please try #2362

PR #2362 (2024-05-15): Only show a singly battery notifcation
From feedback in #2317 it became pretty clear that battery data is a mess with devices providing levels like full, medium, low and devices providing fine-grained percent values, flapping values etc. and we do not want to go into power management applet business, so we only show a single battery notification on connection (when the battery interface shows up or - if the interface is already present when the device connects - when the existing interface sends its first property update) and drop the reference to the connection notification to update so that  [... cut]
commit 46025544 (2024-05-15): Only show a singly battery notifcation

From feedback in #2317 it became pretty clear that battery data is a mess with devices providing levels like full, medium, low and devices providing fine-grained percent values, flapping values etc. and we do not want to go into power management applet business, so we only show a single battery connection (when the battery interface shows up or - if the interface is already present when the device connects - when the existing interface sends its first property update) and drop the reference to the connection notification to update so that further values  [... cut]
PR #2366 (2024-05-15): Only show a singly battery notifcation
Backport of #2362 due to the great demand seen in #2317. This is technically a regression, but the current behavior seems unacceptable in many cases, so let's pretend that we never intentionally added those additional notifications.

Closes #2317
commit 442cfbf0 (2024-05-15): Only show a singly battery notifcation

Backport of #2362 due to the great demand seen in #2317. This is technically a regression, but the current behavior seems unacceptable in many cases, so let's pretend that we never intentionally added those additional notifications.

Closes #2317
closed 2024-05-15 by cschramm