blueman-project/blueman #2318: [2.4] [Bug] Turning on/off adapter (via status menu) doesn't change status switch and devices list

kuraga, 2024-04-03:
blueman: 2.4
BlueZ: 5.72
Distribution: Calculate Linux (Gentoo-based)
Desktop environment: XFCE 4.18

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

1. Start Blueman Applet.
2. Start Blueman Manager.
3. Right-click on the status icon.
4. Click on `Turn Bluetooth Off`.
    * The status icon changed.
    * `Search` toolbar button disabled.
    * `Adapter` menu changed.
    * Devices list _isn't_ changed.
    * Toolbar adapter switch _isn't_ switched off.
5. Close Blueman Manager.
6. Click on the status icon.
    * ...the window is opening for a half-minute...
    * [image]
    * [1]
    * Haven't been able to reproduce _this_ error twice.
    * The window's information is correct.
7. Right-click on the status icon.
8. Click on `Turn Bluetooth On`.
    * The status icon changed.
    * `Search` toolbar button enabled.
    * `Adapter` menu changed.
    * Devices list _isn't_ changed.
    * Toolbar adapter switch _isn't_ switched on.

[1]:
```
blueman-applet 15.16.49 ERROR    DBusProxies:62 activate  : Call to org.blueman.Manager failed
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/DBusProxies.py", line 57, in activate
    self.call_sync("Activate", param, Gio.DBusCallFlags.NONE, -1, None)
gi.repository.GLib.GError: g-io-error-quark: Timeout was reached (24)
```

infirit, 2024-04-03:
For some reason the applet isn't sending the `BluetoothStatusChanged` changed signal the switch relies on to update :disappointed:

kuraga, 2024-04-14:
Saw p.6. in 2.4.1.

PR #2322 (2024-04-03): PowerManager: Add back the BluetoothStatusChanged signal
fixes #2318
commit b9bbf306 (2024-04-03): PowerManager: Add back the BluetoothStatusChanged signal

fixes #2318
commit ef1de08e (2024-04-07): PowerManager: Add back the BluetoothStatusChanged signal

fixes #2318