blueman-project/blueman #2319: [2.4] [Enhancement request] Choose wished device's profile on device connecting

kuraga, 2024-04-03:
blueman: 2.4
BlueZ: 5.72
Distribution: Calculate Linux (Gentoo-based)
Desktop environment: XFCE 4.18

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

The `AudioProfiles` plugin doesn't show any status menu items until you connect to the device.

The `RecentConns` plugin shows _one_ status menu item per device, i.e. _one_ profile per device.

So, I can't choose the wished device's profile on device connecting.

Could it be improved? Thanks!

cschramm, 2024-04-03:
:thinking: Definitely not a super easy change but thinking out loud...

We could remember `(device, audio profile)` tuples and offer them in `RecentConns`. We can obviously not request the device to get connected with the profile active, but we can switch the profile accordingly after the device got picked up by the audio server.

I'm a little unsure about the use case. I do see that one wants to use (legacy?) headsets with different profiles for a) communication and b) listening and that it's a bit cumbersome to first connect the device and then change the profile. Then again, is `RecentConns` really relevant there at all? Isn't the typical flow for headsets that you power them on and they immediately connect to the previously paired device?

kuraga, 2024-04-03:
> I'm a little unsure about the use case.

Turning on the wished profile via _two_ clicks instead of four.
(Thought `AudioProfiles` is exactly for this.)

kuraga, 2024-04-03:
> The `RecentConns` plugin shows _one_ status menu item per device, i.e. _one_ profile per device.

Could change _this_? With a setting, maybe.

cschramm, 2024-04-03:
> > I'm a little unsure about the use case.
> 
> Turning on the wished profile via _two_ clicks instead of four. (Thought `AudioProfiles` is exactly for this.)

Again:

> Isn't the typical flow for headsets that you power them on and they immediately connect to the previously paired device?

It's not a use case for `RecentConns` at all if that's the case.

kuraga, 2024-04-03:
Hm... I have a notebook, a smartphone and headsets.

For the smartphone, the mode is "mixed" every time.
For the notebook, it's either "audio" or "headset".

So, there are three different use cases:
1. Connect to the smartphone.
2. Connect to the notebook as "audio".
3. Connect to the notebook as "headset".

Note there is _no_ the "Connect to the notebook" option.

Don't know whether it's a standard case but want to connect quicker.

kuraga, 2024-04-03:
> It's not a use case for `RecentConns` at all if that's the case.

For a separate, the `RecentConnProfiles` plugin, then?
It will list the recent connected profiles.

cschramm, 2024-04-05:
> > It's not a use case for `RecentConns` at all if that's the case.
> 
> For a separate, the `RecentConnProfiles` plugin, then? It will list the recent connected profiles.

You got me wrong there. I was questioning if a headset is a use case for RecentConns at all as it typically auto-connects when powered on, but of course it is if you use it with multiple devices.

> Also, if `RecentConns` is about "recent connections" (i.e. "devices") not "recent profile-connections", IMHO `<... profiles> on <device>` should be renamed to `Connect to <device>`. The same as we already have `Disconnect <device>`.

Your mixing Bluetooth and audio device profiles. RecentConns is about connections to more or less specific Bluetooth profiles.

kuraga, 2024-04-05:
Auto-connection could help me in half cases only as there are two options (profiles). And auto-connection doesn't know my current wishes.

Also, in most cases, auto-connection enables HSP/HSF profile only fr my headset. But that's another story.
And "auto-connection" is "from the device to the host" here, see https://github.com/blueman-project/blueman/issues/1709#issuecomment-1115324427. (Even if `AutoConnect` plugin is on. The device do it first?)