blueman-project/blueman #2326: V2.4 GTK-Warning

dmillfree868, 2024-04-04:
blueman: 2.4
BlueZ: 5.73
Distribution: Slackware 64 current
Desktop environment: xfce4 4.18

GTK has a warning about markup failing

(blueman-tray:1745): Gtk-WARNING **: 17:17:11.376: Failed to set text 'Audio Gateway (A2DP Source & HSP/HFP AG)' from markup due to error parsing markup: Error on line 1: Entity did not end with a semicolon; most likely you used an ampersand character without intending to start an entity — escape ampersand as &amp;

(blueman-tray:1745): Gtk-WARNING **: 17:17:11.382: Failed to set text '<b>Audio Gateway (A2DP Source & HSP/HFP AG)</b>' from markup due to error parsing markup: Error on line 1: Entity did not end with a semicolon; most likely you used an ampersand character without intending to start an entity — escape ampersand as &amp;

[image]

cschramm, 2024-04-04:
Could you try #2327?

dmillfree868, 2024-04-04:
OK I made the changes to the file PulseAudioProfile.py as suggested logged out and then logged back in, connected the phone and that fixed the issue. Thank You!

PR #2327 (2024-04-04): Escape audio profile names for markup
Closes #2326
commit 33daa953 (2024-04-04): Escape audio profile names for markup

Closes #2326
commit 97351524 (2024-04-07): Escape audio profile names for markup

Closes #2326
closed 2024-04-04 by cschramm