blueman-project/blueman #2328: Regression 2.4: "Recent Connections" is empty

fbuetler, 2024-04-04:
blueman: 2.4-1
BlueZ: 5.73-4
Distribution: arch
Desktop environment: wayland

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Updating from 2.3 to 2.4 introduced a regression having an empty list in the "Recent Connection" tab in the applet. Downgrading to 2.3 showed them again:

2.3:
[image]

2.4:
[image]

Either this or the idea was to remove "Recent Connections" completely as all audio profiles are now inlined?

cschramm, 2024-04-04:
Not sure I can follow. Aren't the items that you blurred your recent connections?

fbuetler, 2024-04-04:
Yes, this possibility is reflected in my last comment

> Either this or the idea was to remove "Recent Connections" completely as all audio profiles are now inlined?

Maybe it is just me not understanding the new UI correctly. For me, the `audio and input profile on ...` look more like configuration options to me. But indeed, clicking on an element there connects the device. 

The main confusion comes from the fact, that in prior versions one had to click on the "Recent Connections" element to connect to a recent device and in 2.4 the element is still there, but its function is now a title (it's also the only title in this menu). 

Maybe this can be reconsidered?

cschramm, 2024-04-04:
We moved the recent connections from the submenu into the toplevel applet menu and "Recent Connections" is a title to give them some context, yes. #1558 / #1652

Prefixing the items with "Connect to..." makes sense, I guess. We could also rename the title to that or "Reconnect to...".

fbuetler, 2024-04-04:
I see you oriented yourself on the nm-applet as they are using similar title styles. In this case, completely dropping the prefix  `Audio and input profile on` might be the slimmest solution, like nm-applet is doing with networks (this makes it even easier as there are no new translations to be added).

cschramm, 2024-04-04:
That's not really an option. E.g. for a phone you typically can have entries like:
* Auto-connect profiles (audio and input)
* Some network profile
* Some serial profile

The auto-connect item will not connect any network or serial profile and vice-versa.

kuraga, 2024-04-05:
> Changing `Audio and input profile on <device>` to `Connect to <device>` might be a possibility.

Came there from another side at https://github.com/blueman-project/blueman/issues/2319#issuecomment-2035353547

kuraga, 2024-04-07:
@fbuetler can you also check #2331, please?

cschramm, 2024-04-09:
Not sure what to do here. I think the main point (ignoring any confusion caused by change itself if you are used to the old menu) is that the "{services} on {device}" items do not immediately tell you that they are about using a recent connection and it does not seem clear from the title and "grouping". Adding "Connect to" to each item would perfectly fix that but be rather bulky. I'd rather try to find a better title like "Reconnect to..." (including the ellipsis in hope that it also makes it more obvious that it's a titel for the following items).

PR #2364 (2024-05-15): Rename recent connections header
I hope this makes things clearer and avoids confusion due to the change from 2.3 to 2.4.

Closes #2328
commit cf53eb7f (2024-05-15): Rename recent connections header (#2364)

I hope this makes things clearer and avoids confusion due to the change from 2.3 to 2.4.

Closes #2328
commit 97eaf6af (2024-09-27): Rename recent connections header (#2364)

I hope this makes things clearer and avoids confusion due to the change from 2.3 to 2.4.

Closes #2328
closed 2024-06-25 by cschramm