blueman-project/blueman #2331: [2.4] [Bug] AudioProfiles plugin doesn't work

kuraga, 2024-04-06:
blueman: 2.4
BlueZ: 5.72
Distribution: Calculate Linux (Gentoo-based)
Desktop environment: XFCE 4.18

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Clicking on the `AudioProfiles` plugin's applet menu items doesn't change anything.

There is no any console output.

(Clicking on the manager context menu items seems to switch the profile normally.)

kuraga, 2024-04-06:
Ok, will check on 2.4.1.

kuraga, 2024-04-07:
[image]

click:

[image]

cschramm, 2024-04-08:
No idea. I'd tell you to gather the values and try the `set_card_profile` call manually, but it's exactly the same as in blueman-manager, so that seems pointless.

You are absolutely sure that `blueman-applet --loglevel debug` does not give any output on the click and it does close the menu as usual? Also try `killall blueman-tray` and run `blueman-tray --loglevel debug` manually.

kuraga, 2024-04-08:
Work after total reboot. Sorry.

kuraga, 2024-04-12:
See it in 2.4.1.

(But, see it after `kill`'ing `blueman-{tray,applet}` and starting on the terminal.
Don't know if it occurs in the wild - re-started to catch another bug.)

kuraga, 2024-04-13:
No, can't get a reproducer.

Closing. But remember this issue :)

kuraga, 2024-04-14:
But it comes again :( Didn't restart.

kuraga, 2024-04-24:
@cschramm ,

https://github.com/blueman-project/blueman/blob/2.4.1/blueman/plugins/applet/Menu.py#L63 doesn't have `'callback'`
=> `dict(item.submenu_items[1])` doesn't have `'callback'`
=> `dict(item)` doesn't have `'callback'`s
=> `self._get_menu()` doesn't have `'callback'`s.

Staudey, 2024-05-09:
I'm seeing this in journalctl when I try to change the audio profile via the applet (on Budgie desktop), in the app itself it's fine:

```
May 09 14:41:57 solus-pc budgie-panel[1908]: DBusMenu.vala:137: Failed to send clicked event to node 2958345: GDBus.Error:org.freedesktop.DBus.Error.Failed: Traceback (most recent call last):
                                               File "/usr/lib/python3.11/site-packages/blueman/main/DbusService.py", line 156, in _handle_method_call
                                                 ok(method(*args))
                                                    ^^^^^^^^^^^^^
                                               File "/usr/lib/python3.11/site-packages/blueman/main/indicators/StatusNotifierItem.py", line 94, in _on_event
                                                 self._on_activate(idx >> 8, idx % (1 << 8) - 1)
                                               File "/usr/lib/python3.11/site-packages/blueman/main/Tray.py", line 62, in activate_menu_item
                                                 AppletService().ActivateMenuItem('(ai)', indexes)
                                               File "/usr/lib/python3.11/site-packages/gi/overrides/Gio.py", line 349, in __call__
                                                 result = self.dbus_proxy.call_sync(self.method_name, arg_variant,
                                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                             gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.Failed: Traceback (most recent call last):
                                               File "/usr/lib/python3.11/site-packages/blueman/main/DbusService.py", line 156, in _handle_method_call
                                                 ok(method(*args))
                                                    ^^^^^^^^^^^^^
                                               File "/usr/lib/python3.11/site-packages/blueman/plugins/applet/Menu.py", line 171, in _activate_menu_item
                                                 node = self.__menuitems[(indexes[0] >> 8, indexes[0] % (1 << 8))]
                                                        ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                                             KeyError: (45, 36)
                                              (0)

```

The menu closes normally after clicking. No output (apart from the starting messages) with `blueman-tray --loglevel debug`

cschramm, 2024-05-09:
Thanks for the data, @Staudey. Good catch and, ouch, I do have an idea what's going on. Checking e.g. `pactl list cards` what card indexes do you see? If I'm right, I'd expect something close to 800.

kuraga, 2024-05-09:
@Staudey ,  is _your_ problem always reproducible?
My is starting at some moment... Until reboot..

Staudey, 2024-05-09:
@kuraga It was indeed not always reproducible, BUT

@cschramm when I checked just now it was only up to index #70 and the issue was no longer reproducible. When I lowered. So I disconnected and reconnected my headphones a bunch of times and when they were at index #270 (i.e. bigger than 255) the issue appeared again.

cschramm, 2024-05-09:
The issue is clear then. I'll look into a fix.

It's probably PipeWire, not genuine pulseaudio, right?

Does it count up or are indexes somehow random? I mean, even if you never properly shut down your system and re-connect your headphones like five to ten times a day, it would still take a couple of months to get to spheres like 800.

kuraga, 2024-05-09:
@cschramm , `1066`, `1132`, `1139`...

Staudey, 2024-05-09:
> It's probably PipeWire, not genuine pulseaudio, right?

Yes, it's pipewire. It does count up with every reconnect, but by a random amount (up to ~50 per step)

closed 2024-05-15 by cschramm

Staudey, 2024-05-15:
I can confirm that the issue is fixed with this patch. Just successfully switched audio profiles through the tray icon when the index of my card was `# 312` and higher. :+1:

PR #2363 (2024-05-15): Fix audio profile menu item priorities
We build a (32 bit) ID for the menu item from its priority values, using the sub priority as the lower 8 bits. Pulseaudio card indexes are theoretically 32 bit and PipeWire re-assignes them with large gaps on re-connect so that it provides high 3-digit numbers in practice that do not fit the 8 bits.

Closes #2331
commit 336e2336 (2024-05-15): Fix audio profile menu item priorities

We build a (32 bit) ID for the menu item from its priority values, using the sub priority as the lower 8 bits. Pulseaudio card indexes are theoretically 32 bit and PipeWire re-assignes them with large gaps on re-connect so that it provides high 3-digit numbers in practice that do not fit the 8 bits.

Closes #2331
commit 69fce8f0 (2024-09-27): Fix audio profile menu item priorities

We build a (32 bit) ID for the menu item from its priority values, using the sub priority as the lower 8 bits. Pulseaudio card indexes are theoretically 32 bit and PipeWire re-assignes them with large gaps on re-connect so that it provides high 3-digit numbers in practice that do not fit the 8 bits.

Closes #2331