blueman-project/blueman #2332: [Bug] Errors on device disconnection

kuraga, 2024-04-06:
blueman: 2.4
BlueZ: 5.72
Distribution: Calculate Linux (Gentoo-based)
Desktop environment: XFCE 4.18

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Clicked `Disconnect device`:

```
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/BatteryWatcher.py", line 27, in <lambda>
    lambda _any_battery, key, value, path: callback(path, value) if key == "Percentage" else None
                                           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/blueman/plugins/applet/ConnectionNotifier.py", line 39, in _on_battery_update
    notification = self._notifications[path]
                   ~~~~~~~~~~~~~~~~~~~^^^^^^
KeyError: '/org/bluez/hci0/dev_1C_6E_4C_80_0D_6E'
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/BatteryWatcher.py", line 27, in <lambda>
    lambda _any_battery, key, value, path: callback(path, value) if key == "Percentage" else None
                                           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/blueman/plugins/applet/ConnectionNotifier.py", line 39, in _on_battery_update
    notification = self._notifications[path]
                   ~~~~~~~~~~~~~~~~~~~^^^^^^
KeyError: '/org/bluez/hci0/dev_1C_6E_4C_80_0D_6E'
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/BatteryWatcher.py", line 27, in <lambda>
    lambda _any_battery, key, value, path: callback(path, value) if key == "Percentage" else None
                                           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/blueman/plugins/applet/ConnectionNotifier.py", line 39, in _on_battery_update
    notification = self._notifications[path]
                   ~~~~~~~~~~~~~~~~~~~^^^^^^
KeyError: '/org/bluez/hci0/dev_1C_6E_4C_80_0D_6E'
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/BatteryWatcher.py", line 27, in <lambda>
    lambda _any_battery, key, value, path: callback(path, value) if key == "Percentage" else None
                                           ^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/blueman/plugins/applet/ConnectionNotifier.py", line 39, in _on_battery_update
    notification = self._notifications[path]
                   ~~~~~~~~~~~~~~~~~~~^^^^^^
KeyError: '/org/bluez/hci0/dev_1C_6E_4C_80_0D_6E'
blueman-applet 11.02.15 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None
blueman-applet 11.02.15 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None
blueman-applet 11.02.15 ERROR    Base:93 callback  : Unhandled error for org.bluez.Device1.Disconnect
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/bluez/Base.py", line 86, in callback
    value = proxy.call_finish(result).unpack()
            ^^^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying (4)
```

BlueZ crashed after this. That's known (https://github.com/bluez/bluez/issues/431) but hadn't seen any Blueman's output on 2.3.5.

kuraga, 2024-04-06:
Also saw on restarting the `bluetooth` service:

```
blueman-applet 15.41.11 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None
blueman-applet 15.41.11 WARNING  DiscvManager:122 update_menuitems: warning: Adapter is None
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/main/Applet.py", line 80, in _on_dbus_name_appeared
    plugin.on_manager_state_changed(self.manager_state)
  File "/usr/lib/python3.11/site-packages/blueman/plugins/applet/AuthAgent.py", line 22, in on_manager_state_changed
    self._agent.register_agent()
  File "/usr/lib/python3.11/site-packages/blueman/main/applet/BluezAgent.py", line 57, in register_agent
    self.register()
  File "/usr/lib/python3.11/site-packages/blueman/main/DbusService.py", line 114, in register
    regid = self._bus.register_object(
            ^^^^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-io-error-quark: An object is already exported for the interface org.bluez.Agent1 at /org/bluez/agent/blueman (2)
```

cschramm, 2024-04-06:
You're probably on Linux 6.8.2. Just up- or downgrade.

closed 2024-04-06 by cschramm

kuraga, 2024-04-06:
> You're probably on Linux 6.8.2. Just up- or downgrade.

Which Linux issue do you mean?

> You're probably on Linux 6.8.2.

```
# uname -a
Linux node-calculate2 6.6.12-calculate #1 SMP PREEMPT_DYNAMIC Wed Jan 17 12:09:30 UTC 2024 x86_64 Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz GenuineIntel GNU/Linux
```

kuraga, 2024-04-06:
By the log above, the adapter variables become cleared before this callbacks fire?

cschramm, 2024-04-06:
See #2304, but 6.6.12 is of course not affected.

We might actually have an issue on disconnection (apart from the fact that those errors are harmless), but #2324 should already fix it.

kuraga, 2024-04-07:
Ok, will check on 2.4.1.

kuraga, 2024-04-24:
@cschramm , on 2.4.1:

```
blueman-applet 18.43.28 ERROR    Base:93 callback  : Unhandled error for org.bluez.Device1.Disconnect
Traceback (most recent call last):
  File "/usr/lib/python3.11/site-packages/blueman/bluez/Base.py", line 86, in callback
    value = proxy.call_finish(result).unpack()
            ^^^^^^^^^^^^^^^^^^^^^^^^^
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.NoReply: Message recipient disconnected from message bus without replying (4)
```