blueman-project/blueman #2344: blueman gui and tray icon doesn't show connections properly

hamkg, 2024-04-14:
blueman: 2.4.1
BlueZ: 5.73
Distribution: slackware 64 -current
Desktop environment: mate, xfce

Don't know when this happened or if it's caused/related by other upgrades, but recently I noticed that even if my BT devices connects and work properly, the tray icon doesn't change colour when connection is active. When opening the gui and choosing Device > Info, it says "Connected no" although it is connected. No popup notifications when connecting/disconnecting a device.

cschramm, 2024-04-14:
You probably need a kernel update.

hamkg, 2024-04-14:
Kernel update? I'm on Linux kernel 6.6.27 - and the icons and connection information has worked earlier.
Perhaps a kernel _downgrade_ might change the behaviour, but I rather not take that path.

infirit, 2024-04-14:
This is a kernel bug, see https://github.com/blueman-project/blueman/issues/2304 for more details. I'm not sure which version of the 6.6 series has the fix but it is in 6.6.25.

cschramm, 2024-04-14:
🤔 I expected the fix in 6.8.5 and 6.6.26 and someone confirmed that the regression is gone for him in 6.8.5, but it seems like the mentioned fix did not make it into any release yet.

hamkg, 2024-04-14:
Thanks a lot. Since my BT devices connect and work OK I'll just wait for a fixed kernel version rather than downgrading.

closed 2024-04-14 by cschramm

cschramm, 2024-04-14:
Double checked now. 6.9 will work and I also expect the fix in 6.8.7 and 6.6.28. My confusion that it's already gone came from Arch Linux maintaining a patch on their side.

hamkg, 2024-04-18:
Fixed in kernel 6.6.28.