blueman-project/blueman #2346: After 2.4, clicking the tray icon no longer minimizes the GUI.

roket1428, 2024-04-20:
Since version 2.4 clicking the tray icon while blueman is visible doesn't minimize it, instead it makes already visible GUI active window. Is this behavior intentional? Downgrading to 2.3.5 reverts to the old behavior where clicking it minimizes the GUI.

blueman: 2.4.+
BlueZ: 5.75
Distribution: Artix Linux
Desktop environment: dwm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I didn't provide any logs since it could be the intended behavior. (will provide if needed)

infirit, 2024-04-20:
For consistency reasons this was changed back to how it was before. You can use the `Ctrl + q` and `Ctrl + w` keyboard shortcuts to close.

roket1428, 2024-04-21:
Any chance of making this an option? I could make this an opt-in feature with a PR if there is any chance of merging it.

Obligatory: [xkcd](https://xkcd.com/1172/)

cschramm, 2024-04-21:
I think the diff should be small enough (see #2122) that a GSettings setting does not hurt.

Why do you want it to hide the window? I do not really have an opinion on the behavior, but the wish to focus the already open window made sense to me for messy desktops.

roket1428, 2024-04-21:
> I think the diff should be small enough (see https://github.com/blueman-project/blueman/pull/2122) that a GSettings setting does not hurt.

Here is the PR: https://github.com/blueman-project/blueman/pull/2347

> Why do you want it to hide the window? I do not really have an opinion on the behavior, but the wish to focus the already open window made sense to me for messy desktops.

I use a window manager, so I can't close windows with a mouse because windows doesn't have a button at the top right to close it, and since I'm using the mouse to interact with the tray icons, switching between keyboard and mouse instead of just using the mouse is much more convenient.

cschramm, 2024-04-23:
Perfectly reasonable as well. I'd even say focusing makes sense on non-tiling window managers and toggling makes sense on tiling ones (at least if they do not have close buttons) then. Ideally we'd detect the window manager and make the behavior depend on it. OK, I'm mostly kidding now. :see_no_evil:

PR #2347 (2024-05-17): Add an option to toggle manager visibility on clicking the tray icon
Fixes #2346
closed 2024-05-17 by infirit