blueman-project/blueman #2348: Headphone audio quality & connection issue

IssamSY, 2024-04-21:
blueman:2.3.5
BlueZ:
Distribution: Ubunto
Desktop environment: Linux Mint 21

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hello everyone,
I submitted this issue regarding a technical problem I face. Whenever in use of my Bluetooth headphone, and connecting it my PC, at some point most of the audio quality drops down. It becomes more like a muted and unclear. 

I tried to see where the problem actually lies on, is it the headphone? No, I confirmed. So, any solution to solve this problem.

The results of running `blueman-applet --loglevel debug`:
[Pastebin](https://pastebin.com/4y1PvHYr)

ADonald63, 2024-06-02:
I have the same issues with abnormal playback, drop outs and volume levels increasing and decreasing and unnatural audio.
Running Bose Comfort 35 headphones
Blueman-Manager 2.3.5

[image]

Any assistance would be greatly appreciated.

infirit, 2024-06-04:
There isn't much blueman can do here. It handles  the initial connection but after that it's up to whatever is handling Bluetooth audio which is PulseAudio or Pipewire.

However, from what you describe pulseaudio/pipewire switched to the headset profile.