blueman-project/blueman #2377: blueman-tray is restarting the xfce4-panel's systray-plugin

7lilnewbie, 2024-05-19:
**blueman**: 2.2.4-1
**BlueZ**: 5.64-0ubuntu1.1
**Distribution**: Zorin 17 Lite (Based on Xubuntu 22.04)
**Desktop environment**: xfce 4.18

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

### Issue
blueman-tray is restarting the xfce4-panel's systray-plugin on most of the toggle on/off.  It's irritating because it removes the bluetooth icon and network icon from the panel for few seconds. It also happens sometimes when I connect to a bluetooth device, upon successful connection/disconnection it restarts the systray-plugin. 

This problem is observed only when the **Appindicator plugin** is enabled in the blueman plugins. If I disable it then it doesn't create any issue but now if I left click on the bluetooth icon then it opens blueman-manager instead of the blueman-tray menu.

### Logs:
 - blueman-tray --loglevel debug: https://termbin.com/ycnp

### Steps to reproduce:
 - Don't know, I did a clean installation a week ago and since then this issue starts happening. 
 - Reboot also doesn't fix this.
 - `killall blueman-tray` get rid of it for few minutes but again it comes back.

### Screen recording:
**Note:** to keep the recording short duration I toggled on/off quickly, but it doesn't matter even if I do it much later the problem still happens.

https://github.com/blueman-project/blueman/assets/145741322/1e9b6939-3ba4-47ed-99c0-12e8abecb56a

cschramm, 2024-05-19:
Unfortunately that's a rather old version of blueman. AppIndicator did get replaced in 2.3. The AppIndicator plugin used libappindicator (or rather libayatana-appindicator in your case) and I'd tend to blame that instead of blueman.

7lilnewbie, 2024-05-21:
> You mind find some useful trace in ~/.xsession-errors or when running xfce4-panel with PANEL_DEBUG=1.

`cat ~/xsession-errors`: https://termbin.com/80n1

Starnge, I had no problem under 2.1.2(in Ubuntu 20.04) even tho it was outdated. Looks like this problem is only in 2.2.4

Since there is no direct way to install the latest blueman as ubuntu jammy repos have only the 2.2.4 version. So I tried to build the 2.4.1 version from the source code but after running `./autorun.sh` stumbled upon an issue:
[image]

It says pygobject package not found but it present:

[image]

I'm not a coding person so don't know how to address this issue, so chatgpt suggested me to find `pygobject-3.0.pc` add it to $PKG_CONFIG_PATH. but I can't locate it where it resides.
[image]

cschramm, 2024-05-21:
I'm not aware of such an issue from Ubuntu 22.04 users either. From the warnings and errors it's pretty clear that the systray plugin crashes. No obvious root cause, but whatever it may be, whether it's in blueman, libayatana-appindicator, libdbusmenu, the xfce4-panel (plugin) or whatever, you most probably will not get any fix for that legacy system anyway and the relevant bits are gone for 3 ½ years on our side now. In case Zorin does receive bug fixes, it's best to file a bug report with them, pointing out that this is not known to be a generic issue with Ubuntu 22.04.

ChatGPT happens to be right and `pygobject-3.0.pc` would be contained in a package called `python-gi-dev`, but you're better off finding a ready-to-use blueman.deb built for your Python 3.10. You could try my snapshots from https://blueman.cschramm.eu/debian/ or the `blueman-git` package from https://launchpad.net/~pipewire-debian/+archive/ubuntu/pipewire-upstream for 2.3.5.

7lilnewbie, 2024-05-22:
> but you're better off finding a ready-to-use blueman.deb built for your Python 3.10. You could try my snapshots from https://blueman.cschramm.eu/debian/

[image]

Thanks, now I got an updated version don't know which one is correct. But it fixed the issue I had, now it doesn't restart the systray-plugin.

But found an other isuue in this updated version, when I click on the **Devices** from the menu it fails to launch the blueman-manager: 
[image]

Logs I got from `blueman-applet --loglevel debug` : https://termbin.com/7iu2
And this error notification appears also:
[image]

But I can launch blueman-manager from the applications menu, it just don't launch from the blueman-tray menu.

7lilnewbie, 2024-05-28:
I came to know that I'm not the only person who gets this D-Bus error. I hope in the future updates it get fixed.

Well, my original problem is solved, so you can close this thread.
Thank you for helping :v:

closed 2024-06-01 by 7lilnewbie